sustainable-computing-io/kepler #1326: daily.yml failed on forked repo

YaSuenag, 2024-04-04:
### What happened?

daily.yml would be invoked everyday even if it is forked repo, but it fails as following:

[image]

### What did you expect to happen?

No workflow error on forked repo.

### How can we reproduce it (as minimally and precisely as possible)?

Fork on GitHub, and wait 1 day :)

### Anything else we need to know?

I think daily.yml should be fired on sustainable-computing.io/kepler only, otherwise image.yml should not be invoked excepts sustainable-computing.io at least.

IMHO all of jobs in daily.yml should be disabled excepts sustainable-computing.io/kepler with `if` statement. What do you think?

### Kepler image tag

<none>

### Kubernetes version

<none>

### Cloud provider or bare metal

<none>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

SamYuan1990, 2024-04-04:
hi @YaSuenag ,
is there any suggestion for the if else conditions to either check if the gha job is running with personal fork?

SamYuan1990, 2024-04-04:
hi @YaSuenag ,
could you please try `GITHUB_ACTION_REPOSITORY`  and `startwith` and submit a PR back?
ref
https://docs.github.com/en/actions/learn-github-actions/variables#default-environment-variables
https://docs.github.com/en/actions/learn-github-actions/expressions

SamYuan1990, 2024-04-04:
I hope some thing like `${{ startsWith(env.GITHUB_ACTION_REPOSITORY, 'sustainable-computing-io') }}` can work.

YaSuenag, 2024-04-04:
I think we can prevent to run the job in following workflow:

```yaml
  image_base:
    if: github.repository == 'sustainable-computing-io/kepler'
    uses: ./.github/workflows/image_base.yml
    secrets:
        username: ${{ secrets.BOT_NAME }}
        password: ${{ secrets.BOT_TOKEN }}
```

I will send PR later.

commit f877482c (2024-04-04): Restrict repository to run daily.yml (#1327)

Closes #1326
closed 2024-04-04 by SamYuan1990