sustainable-computing-io/kepler #1339: Question: Ratio Power Model Is changeable?

SoSonoFuriren, 2024-04-10:
### What would you like to be added?

Hi,
I have a couple of questions regarding Kepler:
- Is it possible to modify the Ratio Power Model in Kepler? Do we have any evidence to support the effectiveness of the Ratio Power Model for distribution?
- Moreover, I also want to know how to package the Kepler source code to an image, if I change the source code of the Kepler.

I would greatly appreciate it if someone could provide answers my questions. Thank you.

### Why is this needed?

Maybe have a better method to distribute the the power of containers.

rootfs, 2024-04-11:
@SoSonoFuriren the ratio model is based on GHG protocol. We can evaluate other protocols/methodologies. 

For packaging, you can find the dockerfile and builder images in the build directory.

SoSonoFuriren, 2024-04-15:
Thanks for your answer!

closed 2024-04-15 by SoSonoFuriren