sustainable-computing-io/kepler #1341: Kepler in Kind - Wrong OPTS for building manifest

juangascon, 2024-04-10:
### Which document would you like to address?

There is mistake in the documentation explaining how to install kepler locally in a Kind cluster
https://sustainable-computing.io/installation/kepler/#running-kepler-on-a-local-kind-cluster
This documentation is rendered from :
https://github.com/sustainable-computing-io/kepler-doc/blob/main/docs/installation/kepler.md

### What is the issue?

In the documentation, it is said that we have to build the Kepler manifest using :
`OPTS="CI_DEPLOY PROMETHEUS_DEPLOY"`.

Though, using this option in the procedure does not work (cf. issue #919 ) because the volume `/proc-host` is not mounted correctly. I know that the issue #919 is about EKS but this happens also with Kind

These are the error messages from the event logs I get on the pods:
> MountVolume.SetUp failed for volume "proc" : hostPath type check failed: /proc-host is not a directory
> Unable to attach or mount volumes: unmounted volumes=[proc], unattached volumes=[], failed to process volumes=[]: timed out waiting for the condition
> MountVolume.SetUp failed for volume "cfm" : failed to sync configmap cache: timed out waiting for the condition
> MountVolume.SetUp failed for volume "redfish" : failed to sync secret cache: timed out waiting for the condition

As it is said in the resolution of issue #919, removing "CI_DEPLOY" from OPTS and leaving only "PROMETHEUS_DEPLOY" makes all work perfectly.
Indeed, when I started using Kepler in its release 0.5, I only used "PROMETHEUS_DEPLOY" to build the manifests and it worked really fine but when I read again the doc for the latest release I thought that "CI_DEPLOY" would add a new feature or evolution.

### How do you suggest this is fixed?

Could you, please, modify the line
`To deploy on a local kind cluster, you need to use the CI_DEPLOY and PROMETHEUS_DEPLOY flags.`
to
`To deploy on a local kind cluster, you need to use the PROMETHEUS_DEPLOY flag.`

and the script line: 
`make build-manifest OPTS="CI_DEPLOY PROMETHEUS_DEPLOY"`
to
`make build-manifest OPTS="PROMETHEUS_DEPLOY"` ?

I do not know if I have to submit instead an issue to the repo where the documentation is rendered from :
https://github.com/sustainable-computing-io/kepler-doc/blob/main/docs/installation/kepler.md

If you prefer, I can fork the documentation repository and propose a pull request with this small change.

Best regards,
Juan GASCON

rootfs, 2024-04-10:
@vprashar2929 @SamYuan1990

SamYuan1990, 2024-04-15:
> ### Which document would you like to address?
> There is mistake in the documentation explaining how to install kepler locally in a Kind cluster [...]

thanks, @juangascon 
I am in sick recently, as you said if it's a small document change. Could you please create a PR for that?
+@sunya-ch , I saw you commented in #919 , could you please help review?

jichenjc, 2024-04-17:
>>>If you prefer, I can fork the documentation repository and propose a pull request with this small change.

@juangascon please open a doc PR for this, I used to create issue here and propose PR there , someone should merge this 
thanks for the finding and report

juangascon, 2024-04-17:
Sorry. I forgot to follow my issue.
@jichenjc I am opening the PR in the doc repo as soon as I end another subject.

juangascon, 2024-04-22:
@SamYuan1990
Thanks for doing the pull request on my behalf. I had some personal issues to manage and could not propose the pull request.
Though, I see in the commit that we are only changing the 'code' block but not the line number 28 explaining the code.
We should also modify it from
To deploy on a local kind cluster, you need to use the CI_DEPLOY and PROMETHEUS_DEPLOY flags.
to
To deploy on a local kind cluster, you need to use the PROMETHEUS_DEPLOY flag.
I could propose a pull request but I do not know if it is going to mess up if I create it before this commit is merged.

closed 2024-04-24 by rootfs