sustainable-computing-io/kepler #1346: kepler_node_core_joules_total=0 on RHEL9/x86_64

jharriga, 2024-04-11:
### What happened?

Downloaded and installed

https://github.com/sustainable-computing-io/kepler/releases/download/v0.7.9/kepler.rpm.tar.gz

On server running
* 5.14.0-417.kpq1.el9.x86_64
* Red Hat Enterprise Linux 9.4 (Plow)

Ran several CPU intensive workloads and metric remained '0'

### What did you expect to happen?

expected the metric reading to increase/track system cpu usage

### How can we reproduce it (as minimally and precisely as possible)?

Download & install rpm
start service
root# systemctl start container-kepler --now
root# curl localhost:8888/metrics | grep

### Kepler image tag

<details>
v0.7.9
</details>

### Kubernetes version

<details>
NONE
</details>

### Cloud provider or bare metal

<details>
bare metal
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
Red Hat Enterprise Linux 9.4 (Plow)

$ uname -a
Linux perf-intel-28.perf.eng.bos2.dc.redhat.com 5.14.0-417.kpq1.el9.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Feb 2 14:05:04 EST 2024 x86_64 x86_64 x86_64 GNU/Linux
</details>

### Install tools

<details>
# rpm --version
RPM version 4.16.1.3
</details>

### Kepler deployment config

<details>
For standalone:
# put your Kepler command argument here
root# systemctl start container-kepler --now
root# curl localhost:8888/metrics | grep
</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2024-05-22:
@jharriga can you double check if it is kepler_node_core_joules_total or kepler_node_package_joules_total?

Current Ampere xgene hwmon only reports the CPU and I/O power (per doc [here](https://docs.kernel.org/hwmon/xgene-hwmon.html)). We cannot get DRAM power. So to align with the RAPL reporting, kepler only reports kepler_node_core_total (per code [here](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/sensors/components/source/apm_xgene_sysfs.go#L72))

jharriga, 2024-06-03:
Running w/v0.7.10 on Ampere server I see:

- kepler_node_core_joules_total{instance="perf-arm-11.perf.eng.bos2.dc.redhat.com",mode="dynamic",package="0",source="intel_rapl"} 98036.551

- kepler_node_package_joules_total{instance="perf-arm-11.perf.eng.bos2.dc.redhat.com",mode="dynamic",package="0",source="intel_rapl"} 98057.08

Both the kepler_node_core_joules_total and kepler_node_package_joules_total metrics do have a values.
This doesn't seem to align with what you expected in the previous comment.

closed 2024-11-19 by rootfs