sustainable-computing-io/kepler #1347: kepler_node_info reports UNKNOWN cpu_architecture on RHEL9/arm64

jharriga, 2024-04-11:
### What happened?

On RHEL9/arm64 system 'kepler_node_info' incorrectly reports “cpu_architecture”

kepler_node_info{components_power_source="ampere-xgene-hwmon",cpu_architecture="unknown",platform_power_source="none",source="os"} 1

### What did you expect to happen?

cpu_architecture indicates the systems cpu arch

# lscpu | grep Model
  Model name:           Neoverse-N1
    BIOS Model name:    Ampere(R) Altra(R) Max Processor

### How can we reproduce it (as minimally and precisely as possible)?

Download & install rpm
start service
root# systemctl start container-kepler --now
root# curl localhost:8888/metrics | grep kepler_node_info

### Kepler image tag

<details>
v0.7.9
</details>

### Kubernetes version

<details>
NONE

</details>

### Cloud provider or bare metal

<details>
bare metal
</details>

### OS version

<details>

# On Linux:
$ cat /etc/os-release
Red Hat Enterprise Linux 9.3 (Plow)

$ uname -a
Linux perf-arm-11.perf.eng.bos2.dc.redhat.com 5.14.0-362.21.1.el9_3.aarch64 #1 SMP PREEMPT_DYNAMIC Thu Jan 25 08:27:11 EST 2024 aarch64 aarch64 aarch64 GNU/Linux

</details>

### Install tools

<details>
# rpm --version
RPM version 4.16.1.3
</details>

### Kepler deployment config

<details>

For standalone:
root# systemctl start container-kepler --now
root# curl localhost:8888/metrics | grep kepler_node_info

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2024-05-22:
current cpu architectures are from `cpu_id`, only working for x86. We need a sysfs id to get for ARM.

jharriga, 2024-07-15:
Seeing same behavior with kepler v0.7.11 on RHEL9/ARM64 Ampere system

[root]# lscpu | grep Model
Model name:                         Neoverse-N1
BIOS Model name:                    Ampere(R) Altra(R) Max Processor

[root]# uname -a
Linux perf-arm-11.perf.eng.bos2.dc.redhat.com 5.14.0-362.21.1.el9_3.aarch64 #1 SMP PREEMPT_DYNAMIC Thu Jan 25 08:27:11 EST 2024 aarch64 aarch64 aarch64 GNU/Linux

[root]# curl localhost:8888/metrics|grep kepler_node_info

kepler_node_info{components_power_source="ampere-xgene-hwmon",**cpu_architecture="unknown"**,platform_power_source="none",source="os"} 1

rootfs, 2024-09-10:
this issue turns out still there in recent code

PR #1482 (2024-06-03): chore(collector)use pmu name to get arm cpu id since archspec does not help here
fix #1347 
we no longer package archspec.