sustainable-computing-io/kepler #1348: Kepler metric readings incorrectly report “source=intel-rapl” on RHEL9/arm64

jharriga, 2024-04-11:
### What happened?

All Kepler metric readings incorrectly report “source=intel-rapl”

- kepler_node_dram_joules_total{instance="perf-arm-11.perf.eng.bos2.dc.redhat.com",mode="dynamic",package="0",source="intel_rapl"} 0

### What did you expect to happen?

source to reflect what mechanism is being used for arm64 systems

Worth noting that 'kepler_node_info' on the same system returns:

- kepler_node_info{components_power_source="ampere-xgene-hwmon",cpu_architecture="unknown",platform_power_source="none",source="os"} 1

### How can we reproduce it (as minimally and precisely as possible)?

Download & install rpm
start service
root# systemctl start container-kepler --now
root# curl localhost:8888/metrics | grep 'kepler-metric'

### Kepler image tag

<details>
v0.7.9
</details>

### Kubernetes version

<details>

NONE

</details>

### Cloud provider or bare metal

<details>
bare metal
</details>

### OS version

<details>

# On Linux:
$ cat /etc/os-release
Red Hat Enterprise Linux 9.3 (Plow)

$ uname -a
Linux perf-arm-11.perf.eng.bos2.dc.redhat.com 5.14.0-362.21.1.el9_3.aarch64 #1 SMP PREEMPT_DYNAMIC Thu Jan 25 08:27:11 EST 2024 aarch64 aarch64 aarch64 GNU/Linux

</details>

### Install tools

<details>
# rpm --version
RPM version 4.16.1.3
</details>

### Kepler deployment config

<details>

For standalone:
root# systemctl start container-kepler --now
root# curl localhost:8888/metrics | grep 'kepler-metric'

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2024-05-22:
this should be fixed in [here](https://github.com/sustainable-computing-io/kepler/pull/1371/files#diff-416e2e8c8782ecce878eb1e637c3d5c70f167b380c63dc21ff1761ef2c524031) and picked up in 0.7.10 release.

closed 2024-05-22 by rootfs

jharriga, 2024-06-03:
Running v0.7.10 on Ampere server and I still see 'source=intel-rapl'

# uname -r
5.14.0-362.21.1.el9_3.aarch64

# dnf list installed | grep kepler
container-kepler.noarch                          0.7.10-1                         @System          

#  curl localhost:8888/metrics | grep kepler_node | wc -l 
21

An example record:
kepler_node_core_joules_total{instance="perf-arm-11.perf.eng.bos2.dc.redhat.com",mode="dynamic",package="0",**source="intel_rapl"**} 97946.062