sustainable-computing-io/kepler #1350: Could not find any ACPI power meter path. Is it a VM?

marvin-steinke, 2024-04-12:
### What happened?

Kepler does not seem to be able to use ACPI.

### What did you expect to happen?

Kepler can use ACPI.

### How can we reproduce it (as minimally and precisely as possible)?

I'm running K3s and installed Kepler with helm according to the [docs](https://sustainable-computing.io/installation/kepler-helm/).

### Anything else we need to know?

Host Specs:
OS: Fedora 39
Host: MS-7A38 8.0 
Kernel: 6.7.9-200.fc39.x86_64 
CPU: AMD Ryzen 7 5800X3D (16) @ 3.400GHz 
GPU: AMD ATI Radeon RX 7900 XT
Memory: 32GB

### Kepler image tag

<details>

```bash
I0412 15:22:42.555299       1 gpu.go:47] Trying to initialize GPU collector using dcgm
W0412 15:22:42.555531       1 gpu_dcgm.go:104] There is no DCGM daemon running in the host: libdcgm.so not Found
W0412 15:22:42.555571       1 gpu_dcgm.go:108] Could not start DCGM. Error: libdcgm.so not Found
I0412 15:22:42.555578       1 gpu.go:54] Error initializing dcgm: not able to connect to DCGM: libdcgm.so not Found
I0412 15:22:42.555580       1 gpu.go:47] Trying to initialize GPU collector using nvidia-nvml
I0412 15:22:42.555633       1 gpu.go:54] Error initializing nvidia-nvml: failed to init nvml. ERROR_LIBRARY_NOT_FOUND
I0412 15:22:42.555637       1 gpu.go:47] Trying to initialize GPU collector using dummy
I0412 15:22:42.555639       1 gpu.go:51] Using dummy to obtain gpu power
I0412 15:22:42.563021       1 exporter.go:155] Kepler running on version: release-0.7.8
I0412 15:22:42.563034       1 config.go:280] using gCgroup ID in the BPF program: true
I0412 15:22:42.563045       1 config.go:282] kernel version: 6.7
I0412 15:22:42.563120       1 config.go:307] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0412 15:22:42.563125       1 exporter.go:167] LibbpfBuilt: true, BccBuilt: false
I0412 15:22:42.563128       1 exporter.go:186] EnabledBPFBatchDelete: true
I0412 15:22:42.563175       1 power.go:53] use sysfs to obtain power
I0412 15:22:42.563181       1 redfish.go:169] failed to get redfish credential file path
I0412 15:22:42.564345       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0412 15:22:42.564352       1 power.go:72] using none to obtain power
I0412 15:22:42.564366       1 exporter.go:201] Initializing the GPU collector
I0412 15:22:42.565008       1 watcher.go:66] Using in cluster k8s config
I0412 15:22:42.665234       1 watcher.go:137] k8s APIserver watcher was started
I0412 15:22:42.665318       1 prometheus_collector.go:92] Registered Container Prometheus metrics
I0412 15:22:42.665345       1 prometheus_collector.go:97] Registered VM Prometheus metrics
I0412 15:22:42.665357       1 prometheus_collector.go:101] Registered Node Prometheus metrics
libbpf: loading /var/lib/kepler/bpfassets/amd64_kepler.bpf.o
libbpf: elf: section(3) kprobe/finish_task_switch, size 2776, link 0, flags 6, type=1
libbpf: sec 'kprobe/finish_task_switch': found program 'kprobe__finish_task_switch' at insn offset 0 (0 bytes), code size 347 insns (2776 bytes)
libbpf: elf: section(4) .relkprobe/finish_task_switch, size 432, link 32, flags 40, type=9
libbpf: elf: section(5) tracepoint/irq/softirq_entry, size 144, link 0, flags 6, type=1
libbpf: sec 'tracepoint/irq/softirq_entry': found program 'kepler_irq_trace' at insn offset 0 (0 bytes), code size 18 insns (144 bytes)
libbpf: elf: section(6) .reltracepoint/irq/softirq_entry, size 16, link 32, flags 40, type=9
libbpf: elf: section(7) kprobe/mark_page_accessed, size 104, link 0, flags 6, type=1
libbpf: sec 'kprobe/mark_page_accessed': found program 'kprobe__mark_page_accessed' at insn offset 0 (0 bytes), code size 13 insns (104 bytes)
libbpf: elf: section(8) .relkprobe/mark_page_accessed, size 16, link 32, flags 40, type=9
libbpf: elf: section(9) kprobe/set_page_dirty, size 104, link 0, flags 6, type=1
libbpf: sec 'kprobe/set_page_dirty': found program 'kprobe__set_page_dirty' at insn offset 0 (0 bytes), code size 13 insns (104 bytes)
libbpf: elf: section(10) .relkprobe/set_page_dirty, size 16, link 32, flags 40, type=9
libbpf: elf: section(11) .data, size 4, link 0, flags 3, type=1
libbpf: elf: section(12) .bss, size 4, link 0, flags 3, type=8
libbpf: elf: section(13) .maps, size 416, link 0, flags 3, type=1
libbpf: elf: section(14) license, size 4, link 0, flags 3, type=1
libbpf: license of /var/lib/kepler/bpfassets/amd64_kepler.bpf.o is GPL
libbpf: elf: section(23) .BTF, size 35505, link 0, flags 0, type=1
[... 208 lines cut ...]
VERSION="39 (Workstation Edition)"
ID=fedora
VERSION_ID=39
VERSION_CODENAME=""
PLATFORM_ID="platform:f39"
PRETTY_NAME="Fedora Linux 39 (Workstation Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:39"
DEFAULT_HOSTNAME="fedora"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f39/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=39
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=39
SUPPORT_END=2024-11-12
VARIANT="Workstation Edition"
VARIANT_ID=workstation

```

</details>

jichenjc, 2024-04-17:
the code is here 
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/sensors/platform/source/acpi.go#L76

guess either AMD is not considered or handle well or the helm chart has some bug
maybe you can follow above code logic and check following as well  to see any clue can be found?

https://github.com/sustainable-computing-io/kepler/blob/main/pkg/sensors/platform/source/acpi.go#L37

marvin-steinke, 2024-04-17:
> the code is here https://github.com/sustainable-computing-io/kepler/blob/main/pkg/sensors/platform/source/acpi.go#L76
>
> guess either AMD is not considered or handle well or the helm chart has some [...]

I see, I don't think there's a bug in the helm charts. At the acpi power path `/sys/devices/LNXSYSTM:00` there are no files that would match. However, `devices/pci0000:00/0000:00:03.1/0000:29:00.0/0000:2a:00.0/0000:2b:00.0/hwmon/hwmon0/power1_average` looks like a file that matches the pre- and suffix. I'm not quite familiar with acpi, would this file be what I'm looking for?  It does contain a single value, right now e.g. 12000000, which is updated each time I reopen.

jichenjc, 2024-04-18:
@rootfs @marceloamaral maybe you can comment on the ACPI question here?

rootfs, 2024-04-18:
This is benign. ACPI is not always on. Please feel free to create a PR to change the logging message.

closed 2024-04-18 by marvin-steinke

kennytobby, 2024-04-18:
Hi, @rootfs. Sorry I have the related problem about the ACPI. If my Kernel of machine can support ACPI, but I can't found the power1_avg in sys/class/hwmon/hwmon2/device path. Do you have any the command can trigger it? Or my some configuration need to revise it?

rootfs, 2024-04-18:
@kennytobby In that case, the `hwmon2` may not be a power meter. The hwmon naming is a bit tricky, some doc can be found [here](https://docs.kernel.org/hwmon/acpi_power_meter.html)

kennytobby, 2024-04-19:
Hi, @rootfs Thank you very much for your reply. So, the the power_path about kepler_node_platform_joules_total from ACPI has 2 ways. One path is /sys/class/hwmon/hwmon2/device/power1_average. the other is /sys/devices/LNXSYSTM:00 , right? And the power meter function is only support in ACPI 4.0? Or if my linux kernel is support ACPI, it will availble in my machine.