sustainable-computing-io/kepler #1360: Use make targets instead of docker actions for building and pushing images on CI

vprashar2929, 2024-04-17:
### What would you like to be added?

A GH workflow responsible for building and pushing images of Kepler using make targets. 

### Why is this needed?

Currently, we use pre-defined actions for building and pushing images for Kepler to quay. In certain situations where images are not built or pushed correctly, it is very hard to investigate the issue. 

Using our existing make targets for building and pushing images has the advantage that it's running the same "make" steps as a developer would run, so less risk of things working differently in CI from in development.

SamYuan1990, 2024-04-17:
in this case, how we ensure multi-arch images support?

4censord, 2024-06-14:
Parametrize the arch in the makefile e.g., `make image ARCH=arm64`.
Then just use a matrix in the action to build for all supported arch values.