sustainable-computing-io/kepler #1405: CPU Ref Cycles in unused

dave-tucker, 2024-05-03:
### What would you like to be added?

In working on #1398 I found that there are a number of unused eBPF maps.

See here: https://github.com/sustainable-computing-io/kepler/blob/main/bpfassets/bcc/bcc.c#L217

That the variable `on_cpu_avg_freq` is not used, therefore the function [`get_on_cpu_avg_freq`](https://github.com/sustainable-computing-io/kepler/blob/main/bpfassets/bcc/bcc.c#L166) and the map `cpu_freq_array`

Following that through, the same is then true of [`on_cpu_ref_cycles_delta`](https://github.com/sustainable-computing-io/kepler/blob/main/bpfassets/bcc/bcc.c#L214), which gets populated by [`get_on_cpu_ref_cycles`](https://github.com/sustainable-computing-io/kepler/blob/main/bpfassets/bcc/bcc.c#L120) and uses the maps `cpu_ref_cycles` and `cpu_ref_cycles_hc_reader`.

In addition, the global variables `HZ` and `CPU_REF_FREQ` are also unused.
N.B - `SAMPLE_RATE` wasn't being correctly populated before #1398 but it is now.

Before I remove all of this unused code it would be excellent to get some clarification on whether it is intended to be used or not.

cc @marceloamaral 

### Why is this needed?

In particular, the perf event array `cpu_ref_cycles_hc_reader` is populated by a hardware perf event and it's my understanding that these are a scarce resource. If we aren't intending to use it, then we should not open a perf event of this type/event.

vimalk78, 2024-05-10:
if we remove collecting cpu freq in bpf side, we can also remove cpu freq related code in user space, and remove the metrics as well.

dave-tucker, 2024-06-04:
Confirmed with @marceloamaral on the community meeting that this is unused and therefore can be removed from both the kernel and userspace side of kepler.

PR #1493 (2024-06-11): chore: Remove CPU Ref Cycles related code
Removes a number of unused maps from the eBPF code.
Removes unused functions and config options from the userland code.

Fixes: #1405
commit bef22c55 (2024-06-11): chore: Remove CPU Ref Cycles related code

Removes a number of unused maps from the eBPF code.
Removes unused functions and config options from the
userland code.

Fixes: #1405
closed 2024-06-11 by sthaha