sustainable-computing-io/kepler #1422: Kepler rpm not installable

aspanner, 2024-05-13:
### What happened?

Tried to install the latest RPM but it was only 6.4kByte of size, reverted back to earlier version (size 8.4MB) and it installed OK:
https://github.com/sustainable-computing-io/kepler/releases/tag/v0.7.9
[image]

### What did you expect to happen?

https://github.com/sustainable-computing-io/kepler/releases/tag/v0.7.9 rpm should install normally

BTW the tag says 0.7.9 but the version is named 0.7.8

### How can we reproduce it (as minimally and precisely as possible)?

Follow the 'download' link from the kepler doc:
https://sustainable-computing.io/installation/kepler-rpm/

Try to install rpm tar.gz from https://github.com/sustainable-computing-io/kepler/releases/tag/v0.7.9

### Kepler image tag

https://github.com/sustainable-computing-io/kepler/releases/tag/v0.7.9

### Kubernetes version

n/a - rpm version

### Cloud provider or bare metal

virtual machine

### OS version

<details>

```console
$ cat /etc/os-release RHEL 9.2
NAME="Red Hat Enterprise Linux"
VERSION="9.2 (Plow)"
ID="rhel"
ID_LIKE="fedora"
VERSION_ID="9.2"
PLATFORM_ID="platform:el9"
PRETTY_NAME="Red Hat Enterprise Linux 9.2 (Plow)"
ANSI_COLOR="0;31"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:redhat:enterprise_linux:9::baseos"
HOME_URL="https://www.redhat.com/"
DOCUMENTATION_URL="https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/9"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 9"
REDHAT_BUGZILLA_PRODUCT_VERSION=9.2
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="9.2"

$ uname -a
Linux localhost.localdomain 5.14.0-284.11.1.el9_2.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Apr 12 10:45:03 EDT 2023 x86_64 x86_64 x86_64 GNU/Linux

</details>

### Install tools

<details>

</details>

### Kepler deployment config

n/a

### Container runtime (CRI) and version (if applicable)

n/a rpm install no container

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2024-05-15:
@aspanner 0.7.9 was a bad build. we'll have a new build this week @vprashar2929 @sthaha

rootfs, 2024-05-16:
@aspanner 0.7.10 is there now. Note the current rpm release is basically a systemd unit that starts podman container. 

```bash
# wget https://github.com/sustainable-computing-io/kepler/releases/download/v0.7.10/kepler.rpm.tar.gz
# tar xvzf kepler.rpm.tar.gz
# yum install RPMS/noarch/container-kepler-0.7.10-1.noarch.rpm
# systemctl enable container-kepler --now
```
```console
# verify podman starts kepler
# podman ps
CONTAINER ID  IMAGE                                           COMMAND     CREATED         STATUS         PORTS       NAMES
111262d94303  quay.io/sustainable_computing_io/kepler:latest              15 minutes ago  Up 16 minutes              kepler

# verify kepler is working

# curl localhost:8888/metrics|grep kepler_node_package_joules_total

```

rootfs, 2024-05-16:
@aspanner can you verify the steps and update the kepler-doc? thanks!

aspanner, 2024-05-16:
@rootfs @sthaha @vprashar2929 - I am on it.