sustainable-computing-io/kepler #1427: Add workflow to build and push Kepler builder image

vprashar2929, 2024-05-14:
### What would you like to be added?

A step or worklfow that builds and push the Kepler builder image so that it shouldn't be done manually

### Why is this needed?

Currently, there is no workflow that builds and pushes the Kepler builder image

vprashar2929, 2024-06-25:
@SamYuan1990 I was looking at how currently the builder image is build and pushed and came across [image_base.yml ](https://github.com/sustainable-computing-io/kepler/blob/main/.github/workflows/image_base.yml)workflow which I think is used in[ daily.yml](https://github.com/sustainable-computing-io/kepler/blob/main/.github/workflows/daily.yml). Can you share the idea behind why we added this workflow in daily and not on push to main branch?

SamYuan1990, 2024-06-25:
> @SamYuan1990 I was looking at how currently the builder image is build and pushed and came across [image_base.yml [...]

we don't change it in the most of time, so seems no need for push at each time.
we'd better build ti daily to catch up CVE if possible. as https://github.com/sustainable-computing-io/kepler/commit/20385219b8230fad9c6d1c792b57cde4c1419292#diff-1fb6f6d52a74b1f02eafeeb5d7649bdee7fadc8fdb1ff7170d58467762188212R30 we have yum update once up on a time.

dave-tucker, 2024-06-25:
I'm hoping that after https://github.com/sustainable-computing-io/kepler/issues/1566 we can remove the builder image entirely 🙏

SamYuan1990, 2024-06-26:
> I'm hoping that after #1566 we can remove the builder image entirely 🙏

I suppose there different parts of builder image's responsibility.
- for RPM build, I suppose we can make it anywhere.... even if a temp container image.
- https://github.com/sustainable-computing-io/kepler/blob/main/build/Dockerfile#L53-L58 as ref to some best practices for container build. a builder image for necessary dependencies and just keep binary in "product" image.

to remove builder image... we need to ....
I suppose at least, we will have a golang image as builder to build binary?

dave-tucker, 2024-06-26:
> I suppose there different parts of builder image's responsibility.
>
> * for RPM build, I suppose we can make it anywhere.... even if a temp container image.
> * [...]

I believe with the changes mentioned above we can just use the RHEL UBI 9 golang image without any additional customization.

SamYuan1990, 2024-06-29:
> I believe with the changes mentioned above we can just use the RHEL UBI 9 golang image without any additional customization.

emmmm, it works for golang, but pre @rootfs told me, we want to have ebpf inside container image. how about ebpf builders like gcc/make?