sustainable-computing-io/kepler #1449: Expose Kepler version in exported metric

vprashar2929, 2024-05-19:
### What would you like to be added?

Currently, Kepler exports a metric `kepler_exporter_build_info`. It also adds a label `Version` but no value is assigned to it. 

### Why is this needed?

It would be beneficial to have the Kepler version, such as `0.7.10`, available as a metric label. This would enable verification of the specific version of Kepler being deployed.

sunya-ch, 2024-05-24:
@vprashar2929 I'm also having an idea to export also the other information in https://github.com/sustainable-computing-io/kepler/issues/1449.
I think there is no need to be collected by prometheus since it will never change after deployment. I think we can extend the Kepler REST API to report these information values. With this API, Kepler-operator can also reuse and update to the Kepler CR status. 

cc: @sthaha @rootfs

jichenjc, 2024-05-24:
>>>I'm also having an idea to export also the other information in https://github.com/sustainable-computing-io/kepler/issues/1449

@rootfs seems you are pointing to this issue, guess it might be some other issue?

sthaha, 2024-05-28:
> I think there is no need to be collected by prometheus since it will never change after deployment. 

@sunya-ch , I actually think this is a  require information especially for long running clusters where kepler gets upgraded from version to version. I will be quite useful to users to understand the version of kepler running on each node. 

E.g. You can see this info exported by prometheus 
```
prometheus_build_info{branch="main", goarch="amd64", goos="linux", goversion="go1.22.3", instance="localhost:9090", job="prometheus", revision="707e9d917eb16c583a79c38891bc8209e29ed983", tags="netgo,builtinassets,stringlabels", version="2.52.0-rc.1"}
```