sustainable-computing-io/kepler #1502: Consider enforcing Conventional Commits

dave-tucker, 2024-06-07:
### What would you like to be added?

See: https://www.conventionalcommits.org/en/v1.0.0/

### Why is this needed?

With this, tooling could be used that performs automatic changelog generation etc.. like https://github.com/googleapis/release-please

vprashar2929, 2024-06-28:
@dave-tucker I was looking at the current [workflow](https://github.com/sustainable-computing-io/kepler/blob/main/.github/workflows/commitMsg.yml) and it doesn't do justice in terms of checking the conventional commits. I was looking at [this](https://github.com/webiny/action-conventional-commits) and believe this would work. Wdyt?

vprashar2929, 2024-07-08:
closing as #1596 is merged

closed 2024-07-08 by vprashar2929