sustainable-computing-io/kepler #1530: kepler_process_bpf_cpu_time_ms_total metrics turned to 0 in latest code

rootfs, 2024-06-13:
### What happened?

We need BPF CPU time to estimate dynamic power on VM. But the metrics are recently unstable.

# In release-0.7.8, the kepler_process_bpf_cpu_time_ms_total is correct:
```console
kepler_process_bpf_cpu_time_ms_total{command="systemd",container_id="system_processes",pid="188268",source="bpf",vm_id=""} 2
kepler_process_bpf_cpu_time_ms_total{command="sshd",container_id="system_processes",pid="592344",source="bpf",vm_id=""} 5
kepler_process_bpf_cpu_time_ms_total{command="pmdalinux",container_id="system_processes",pid="1478",source="bpf",vm_id=""} 6
kepler_process_bpf_cpu_time_ms_total{command="redis-server",container_id="system_processes",pid="1281",source="bpf",vm_id=""} 7
kepler_process_bpf_cpu_time_ms_total{command="pmproxy",container_id="system_processes",pid="4020821",source="bpf",vm_id=""} 9
kepler_process_bpf_cpu_time_ms_total{command="pmdaproc",container_id="system_processes",pid="1476",source="bpf",vm_id=""} 13
kepler_process_bpf_cpu_time_ms_total{command="kworker/4:1",container_id="kernel_processes",pid="590323",source="bpf",vm_id=""} 7082
```

# In release-0.7.10, all the CPU time goes to pid 0:
```console
kepler_process_bpf_cpu_time_ms_total{command="etcd",container_id="system_processes",pid="996366",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="in:imjournal",container_id="system_processes",pid="1299",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="284272",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="284297",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="prometheus",container_id="system_processes",pid="512773",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="swapper/0",container_id="kernel_processes",pid="0",source="bpf",vm_id=""} 1203
```

# In latest code, all cpu time is zero
```console
kepler_process_bpf_cpu_time_ms_total{command="pmie",container_id="system_processes",pid="1672428",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmlogger",container_id="system_processes",pid="3444157",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmlogger",container_id="system_processes",pid="3444480",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmproxy",container_id="system_processes",pid="4020821",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="redis-server",container_id="system_processes",pid="1281",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="systemd-journal",container_id="system_processes",pid="761",source="bpf",vm_id=""} 0
```

### What did you expect to happen?

We need BPF CPU time to estimate dynamic power on VM. But the metrics are recently unstable.

### How can we reproduce it (as minimally and precisely as possible)?

Start container-kepler service on RHEL9 and run the following:
```
curl localhost:8888/metrics | grep kepler_process_bpf_cpu_time_ms_total |sort -k 2 -g
```

### Kepler image tag

<details>
0.7.8, 0.7.10, and latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```
None
</details>

### Cloud provider or bare metal

<details>
bare metal
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE}
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE}
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

marceloamaral, 2024-06-14:
@rootfs could you please share the Kepler logs of the latest code?

dave-tucker, 2024-06-14:
Sharing my local testing here. Running Kepler on Fedora 40 with:

```console
sudo ENABLE_PROCESS_METRICS=true ./_output/bin/linux_amd64/kepler
```

I see some processes that are 0, but plenty that are non-zero:

```console
$ curl localhost:8888/metrics | grep kepler_process_bpf_cpu_time_ms_total |sort -k 2 -g
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="442204",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="prometheus",container_id="39531a28d61d45452f2861f02fc3552c2e06a7496e31962ec92202c585d2447e",pid="3574948",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="kubelet",container_id="system_processes",pid="3571585",source="bpf",vm_id=""} 0.045
kepler_process_bpf_cpu_time_ms_total{command="sshd",container_id="system_processes",pid="441843",source="bpf",vm_id=""} 0.06
kepler_process_bpf_cpu_time_ms_total{command="containerd",container_id="system_processes",pid="3569041",source="bpf",vm_id=""} 0.197
kepler_process_bpf_cpu_time_ms_total{command="kubelet",container_id="system_processes",pid="3569914",source="bpf",vm_id=""} 0.295
kepler_process_bpf_cpu_time_ms_total{command="grafana-server",container_id="6bc252d0b5144c72b556df201bff50b76d3ff9c92b1e64070cf6e1d306676e9d",pid="3575172",source="bpf",vm_id=""} 1.081
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="442190",source="bpf",vm_id=""} 1.59
kepler_process_bpf_cpu_time_ms_total{command="zsh",container_id="system_processes",pid="504330",source="bpf",vm_id=""} 4.319
kepler_process_bpf_cpu_time_ms_total{command="kworker/u64:5",container_id="kernel_processes",pid="482503",source="bpf",vm_id=""} 36736.108
```

rootfs, 2024-06-14:
@marceloamaral here is the log and metrics. Note there is a fraction of cpu time but it is not right

```console
# curl localhost:8888/metrics |grep bpf_cpu_time |sort -k 2 -g
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  101k    0  101k    0     0  49.6M      0 --:--:-- --:--:-- --:--:-- 49.6M
# HELP kepler_container_bpf_cpu_time_ms_total Aggregated value in bpf_cpu_time_ms value from bpf
# HELP kepler_process_bpf_cpu_time_ms_total Aggregated value in bpf_cpu_time_ms value from bpf
# TYPE kepler_container_bpf_cpu_time_ms_total counter
# TYPE kepler_process_bpf_cpu_time_ms_total counter
kepler_container_bpf_cpu_time_ms_total{container_id="system_processes",container_name="system_processes",container_namespace="system",pod_name="system_processes",source="bpf"} 0
kepler_process_bpf_cpu_time_ms_total{command="(systemd)",container_id="system_processes",pid="1437591",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="auditd",container_id="system_processes",pid="906",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="bash",container_id="system_processes",pid="1533232",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="chronyd",container_id="system_processes",pid="940",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="code-89de5a8d4d",container_id="system_processes",pid="1533274",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="coredns",container_id="system_processes",pid="997911",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="coredns",container_id="system_processes",pid="997933",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="etcd",container_id="system_processes",pid="996366",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="grafana-server",container_id="system_processes",pid="1258",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="kindnetd",container_id="system_processes",pid="997347",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="kube-proxy",container_id="system_processes",pid="997230",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="manager",container_id="system_processes",pid="998644",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="manager",container_id="system_processes",pid="998839",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="manager",container_id="system_processes",pid="998912",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="1533315",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="1533359",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="1533370",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="1533387",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmcd",container_id="system_processes",pid="1471",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmdalinux",container_id="system_processes",pid="1478",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmdaproc",container_id="system_processes",pid="1476",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmie",container_id="system_processes",pid="1672189",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmie",container_id="system_processes",pid="1672428",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmlogger",container_id="system_processes",pid="3444157",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmlogger",container_id="system_processes",pid="3444480",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="pmproxy",container_id="system_processes",pid="1026281",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="redis-server",container_id="system_processes",pid="1281",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="rhsmcertd",container_id="system_processes",pid="1024",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="sshd",container_id="system_processes",pid="1533228",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="systemd-journal",container_id="system_processes",pid="761",source="bpf",vm_id=""} 0
kepler_process_bpf_cpu_time_ms_total{command="systemd-logind",container_id="system_processes",pid="945",source="bpf",vm_id=""} 0
kepler_container_bpf_cpu_time_ms_total{container_id="kernel_processes",container_name="kernel_processes",container_namespace="kernel",pod_name="kernel_processes",source="bpf"} 0.435
kepler_process_bpf_cpu_time_ms_total{command="kworker/u32:2",container_id="kernel_processes",pid="461311",source="bpf",vm_id=""} 0.435
# podman ps
CONTAINER ID  IMAGE                                                                                           COMMAND     CREATED       STATUS       PORTS                      NAMES
af8df9491396  docker.io/kindest/node@sha256:047357ac0cfea04663786a612ba1eaba9702bef25227a794b52890dd8bcd692e              9 days ago    Up 9 days    127.0.0.1:34371->6443/tcp  kind-control-plane
[... cut ...]
I0613 15:44:12.560469  594556 prometheus_collector.go:90] Registered Process Prometheus metrics
I0613 15:44:12.560488  594556 prometheus_collector.go:95] Registered Container Prometheus metrics
I0613 15:44:12.560497  594556 prometheus_collector.go:100] Registered VM Prometheus metrics
I0613 15:44:12.560507  594556 prometheus_collector.go:104] Registered Node Prometheus metrics
I0613 15:44:12.560584  594556 exporter.go:201] starting to listen on 0.0.0.0:8888
I0613 15:44:12.560791  594556 exporter.go:215] Started Kepler in 41.964788ms
```

rootfs, 2024-06-14:
In fact, most of the BPF metrics (cpu time, cpu instructions, etc) are now becoming 0. Only pid 0 or one single kernel pid get all the metrics. 

I checked 0.7.2 (our last version of using sched_switch tracepoint), it gets the same symptom. 0.7.3 (used kprobe) and onwards till 0.7.8 work right to me (i.e. all active processes have bpf metrics).

rootfs, 2024-06-14:
The root cause appears to be the wrong padding.

On my RHEL 9, here is the sched_switch args:
```console
# cat /sys/kernel/tracing/events/sched/sched_switch/format 
name: sched_switch
ID: 317
format:
        field:unsigned short common_type;       offset:0;       size:2; signed:0;
        field:unsigned char common_flags;       offset:2;       size:1; signed:0;
        field:unsigned char common_preempt_count;       offset:3;       size:1; signed:0;
        field:int common_pid;   offset:4;       size:4; signed:1;
        field:unsigned char common_preempt_lazy_count;  offset:8;       size:1; signed:0;

        field:char prev_comm[TASK_COMM_LEN];    offset:12;      size:16;        signed:1;
        field:pid_t prev_pid;   offset:28;      size:4; signed:1;
        field:int prev_prio;    offset:32;      size:4; signed:1;
        field:long prev_state;  offset:40;      size:8; signed:1;
        field:char next_comm[TASK_COMM_LEN];    offset:48;      size:16;        signed:1;
        field:pid_t next_pid;   offset:64;      size:4; signed:1;
        field:int next_prio;    offset:68;      size:4; signed:1;

print fmt: "prev_comm=%s prev_pid=%d prev_prio=%d prev_state=%s%s ==> next_comm=%s next_pid=%d next_prio=%d", REC->prev_comm, REC->prev_pid, REC->prev_prio, (REC->prev_state & ((((0x0000 | 0x0001 | 0x0002 | 0x0004 | 0x0008 | 0x0010 | 0x0020 | 0x0040) + 1) << 1) - 1)) ? __print_flags(REC->prev_state & ((((0x0000 | 0x0001 | 0x0002 | 0x0004 | 0x0008 | 0x0010 | 0x0020 | 0x0040) + 1) << 1) - 1), "|", { 0x0001, "S" }, { 0x0002, "D" }, { 0x0004, "T" }, { 0x0008, "t" }, { 0x0010, "X" }, { 0x0020, "Z" }, { 0x0040, "P" }, { 0x0080, "I" }) : "R", REC->prev_state & (((0x0000 | 0x0001 | 0x0002 | 0x0004 | 0x0008 | 0x0010 | 0x0020 | 0x0040) + 1) << 1) ? "+" : "", REC->next_comm, REC->next_pid, REC->next_prio
```

The paddings are 12 bytes.

After I applied the right padding, the cpu time is back
```
kepler_process_bpf_cpu_time_ms_total{command="sshd",container_id="system_processes",pid="2444904",source="bpf",vm_id=""} 0.04
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="2445013",source="bpf",vm_id=""} 0.096
kepler_process_bpf_cpu_time_ms_total{command="prometheus",container_id="9421e5dd315b98cf271ccbb060f55731a9d484d77a796d756e587746c0490983",pid="17376",source="bpf",vm_id=""} 0.097
kepler_process_bpf_cpu_time_ms_total{command="etcd",container_id="274c41da423ac675bdd6d44d1204ada79e4aaa0ea4908dfb69991c6315409585",pid="12013",source="bpf",vm_id=""} 0.101
kepler_process_bpf_cpu_time_ms_total{command="systemd",container_id="system_processes",pid="10035",source="bpf",vm_id=""} 0.218
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="2445057",source="bpf",vm_id=""} 0.556
kepler_process_bpf_cpu_time_ms_total{command="node",container_id="system_processes",pid="2444188",source="bpf",vm_id=""} 0.56
kepler_process_bpf_cpu_time_ms_total{command="gsd-housekeepin",container_id="system_processes",pid="10492",source="bpf",vm_id=""} 1.652
kepler_process_bpf_cpu_time_ms_total{command="ps",container_id="system_processes",pid="2877291",source="bpf",vm_id=""} 10.911
kepler_process_bpf_cpu_time_ms_total{command="kubelet",container_id="system_processes",pid="2211",source="bpf",vm_id=""} 11.975
kepler_process_bpf_cpu_time_ms_total{command="adapter",container_id="55decdb48c42e4592573692fcd287a2a4755ece0ed4d3230edbedda277bd2ee4",pid="17216",source="bpf",vm_id=""} 26.682
kepler_process_bpf_cpu_time_ms_total{command="gvfsd-trash",container_id="system_processes",pid="11157",source="bpf",vm_id=""} 39.622
kepler_process_bpf_cpu_time_ms_total{command="gvfs-udisks2-vo",container_id="system_processes",pid="10414",source="bpf",vm_id=""} 56.6
kepler_process_bpf_cpu_time_ms_total{command="swapper/2",container_id="kernel_processes",pid="0",source="bpf",vm_id=""} 12480.942
```

Here is the diff:
```
# git diff
diff --git a/bpfassets/libbpf/src/kepler.bpf.c b/bpfassets/libbpf/src/kepler.bpf.c
index cac1923c..ea764931 100644
--- a/bpfassets/libbpf/src/kepler.bpf.c
+++ b/bpfassets/libbpf/src/kepler.bpf.c
@@ -197,8 +197,11 @@ static inline void collect_metrics_and_reset_counters(
 // This struct is defined according to the following format file:
 // /sys/kernel/tracing/events/sched/sched_switch/format
 struct sched_switch_info {
-       /* The first 8 bytes is not allowed to read */
-       u64 pad;
+       unsigned short common_type;   
+    unsigned char common_flags;  
+    unsigned char common_preempt_count;   
+    int common_pid; 
+    unsigned char common_preempt_lazy_count; 
 
        char prev_comm[16];
        pid_t prev_pid;
```

rootfs, 2024-06-14:
@dave-tucker @marceloamaral can you check the paddings on your env? If the paddings are different, we should switch back to kprobe for consistency.

dave-tucker, 2024-06-15:
> @dave-tucker @marceloamaral can you check the paddings on your env? If the paddings are different (e.g. as [this](https://android.googlesource.com/platform/system/bpfprogs/+/482dfa1ca63eb209866ff3a7b3aeb3daada7b4e1/time_in_state.c#47) one), we should switch back to kprobe for consistency.

The point of tracepoints is that the ABI is stable. kprobes do not have a stable ABI and therefore could change between kernel versions.

Let me look into this and see what options we have.

dave-tucker, 2024-06-17:
@rootfs ok, so I've got 2 options from you:

- #1543 changes less code and solves the problem in a portable manner
- #1542 brings us to a more modern (Kernel 5.5+) means of using tracepoints which not only fixes the issue, but it is also supposed to be faster

PR #1542 (2024-06-17): fix(bpf): Use BTF-Defined Raw Tracepoints
It turns out that RHEL is carrying a patch that increases the size of the unreadable tracepoint-common area from 8 bytes to 12 bytes. This means that our reads into the context area are off, causing the probes to return invalid data.

To fix this we use BTF-Defined Raw Tracepoints that were added in Kernel 5.5 (less than our 5.12 minimum version). This type of tracepoint program uses BTF and CO-RE to access arguments from the tracepoint context.

Fixes: #1530
closed 2024-06-17 by rootfs