sustainable-computing-io/kepler #1533: Documenting Kepler Metrics Validation framework

husky-parul, 2024-06-14:
### Which document would you like to address?

- Design document of Kepler Validation Framework
- README on getting started
- Explanation on choice of metrics
- Explanation of queries

### What is the issue?

Missing information on how to get started.

### How do you suggest this is fixed?

Needs documentation

husky-parul, 2024-06-20:
### Question about the model weights + validation
At the moment I see we use the weights from here -> [data/model_weight](https://github.com/sustainable-computing-io/kepler/tree/main/data/model_weight) inside the validator module. Kepler 0.7 models are present here https://github.com/sustainable-computing-io/kepler-model-db/tree/main/models/v0.7
But I dont see the models used in[ data/model_weight ](https://github.com/sustainable-computing-io/kepler/tree/main/data/model_weight)present in the https://github.com/sustainable-computing-io/kepler-model-db/tree/main/models/v0.7

- How did these models in [data/model_weight](https://github.com/sustainable-computing-io/kepler/tree/main/data/model_weight) came about/selected (or why)?
- Are we planning to publish validation reports for  SPECPower and EC2 trained models or we just stick to [data/model_weight](https://github.com/sustainable-computing-io/kepler/tree/main/data/model_weight) . If yes, it seems we need to keep them updated as well?

@sthaha @rootfs @dave-tucker

dave-tucker, 2024-06-20:
Bad link in the PR description made me get the dates wrong 😓 
These are the ones that are currently in data/model_weight https://github.com/sustainable-computing-io/kepler-model-db/pull/16

sunya-ch, 2024-06-20:
> ### Question about the model weights + validation
> At the moment I see we use the weights from here -> [data/model_weight](https://github.com/sustainable-computing-io/kepler/tree/main/data/model_weight) inside the validator module. Kepler 0.7 models are present here https://github.com/sustainable-computing-io/kepler-model-db/tree/main/models/v0.7 But I dont see the models used in[ data/model_weight ](https://github.com/sustainable-computing-io/kepler/tree/main/data/model_weight)present in the https://github.com/sustainable-computing-io/kepler-model-db/tree/main/models/v0.7
> 
> * How did these models in [data/model_weight](https://github.com/sustainable-computing-io/kepler/tree/main/data/model_weight) came about/selected (or why)?
> * Are we planning to publish validation reports for  SPECPower and EC2 trained models or we just stick to [data/model_weight](https://github.com/sustainable-computing-io/kepler/tree/main/data/model_weight) . If yes, it seems we need to keep them updated as well?
> 
> @sthaha @rootfs @dave-tucker @KaiyiLiu1234

The model weight comes from node-type 0 of the latest trained power models uploaded to the kepler-model-db. 
This is added for a default for now but will be removed soon. It used to be applied for the case when we have no network connection to download the model weight from db. 
From our previous meeting long ago, we have an idea to use config map to dynamically give a proper power model instead. 

For the case when we have the network, we suppose to add the logic to select the model from the db dynamically. I created some related issue here: 
- https://github.com/sustainable-computing-io/kepler-model-server/issues/216
- https://github.com/sustainable-computing-io/kepler-model-server/issues/231

We have a discussion to have a CI to retrain and push the new power model when Kepler code changes in the issue. 
- https://github.com/sustainable-computing-io/kepler-model-server/issues/212.