sustainable-computing-io/kepler #1540: Version incorrectly reported in the container images

dave-tucker, 2024-06-14:
### What happened?

If you run `quay.io/sustainable_computing_io/kepler:latest` and then query the version:

```console
$ curl -s localhost:8888/metrics | grep build_info
# HELP kepler_exporter_build_info A metric with a constant '1' value labeled by version, revision, branch, os and arch from which kepler_exporter was built.
# TYPE kepler_exporter_build_info gauge
kepler_exporter_build_info{arch="amd64",branch="main",os="linux",revision="92caca25cbedc5a0c2d6491a1bb017e0996ac654",version="1.21.10"} 1
```

### What did you expect to happen?

1.21.10 is the Go version (probably).

I expected:

```console
curl -s localhost:8888/metrics | grep build_info
# HELP kepler_exporter_build_info A metric with a constant '1' value labeled by version, revision, branch, os and arch from which kepler_exporter was built.
# TYPE kepler_exporter_build_info gauge
kepler_exporter_build_info{arch="amd64",branch="main",os="linux",revision="9212768177df078ed88448638c3e2596777d3615",version="v0.7.10-109-g92127681"} 1
```

Which is the behaviour on the `main` branch

### How can we reproduce it (as minimally and precisely as possible)?

Run the latest kepler image and run the curl commands above

### Kepler image tag

<details>
latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE}
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE}
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

vprashar2929, 2024-06-25:
@dave-tucker Are we good to close this?

dave-tucker, 2024-06-25:
@vprashar2929 yes! Just pulled the latest image and verified the version is now correct. Thanks.

```
# HELP kepler_exporter_build_info A metric with a constant '1' value labeled by version, revision, branch, os and arch from which kepler_exporter was built.
kepler_exporter_build_info{arch="amd64",branch="main",os="linux",revision="a51fa5df844120fb0f49934791f6c28ea0a40c7a",version="v0.7.10-132-ga51fa5df"} 1
```

closed 2024-06-25 by dave-tucker