sustainable-computing-io/kepler #1544: Add a pre-commit checks and CI pipeline for the e2e/validator code

dave-tucker, 2024-06-17:
### What would you like to be added?

A CI pipeline and pre-commit checks that:

- checks formatting
- runs tests

for the e2e/validator code

### Why is this needed?

Because as it stands running `hatch fmt` against the code shows a lot of issues.
It will make formatting consistent and the code much easier to review if we have some pre-commit testing and CI.

dave-tucker, 2024-06-17:
Example pre-commit config:
```
repos:
  - repo: local
    hooks:
      - id: format
        name: format
        entry: hatch fmt
        language: system
        types: [python]
        pass_filenames: false
        verbose: true
```

SamYuan1990, 2024-06-17:
what's `hatch fmt`?

dave-tucker, 2024-06-17:
> what's `hatch fmt`?

See: https://hatch.pypa.io/latest/cli/reference/#hatch-fmt

Looks like whomever contributed this code scaffolded it with `hatch` so it probably makes sense to stick with it.
I haven't worked with Python in a very long time so I have no opinions on whether this is right/wrong or best practice.

SamYuan1990, 2024-06-17:
> See: https://hatch.pypa.io/latest/cli/reference/#hatch-fmt
>
> Looks like whomever contributed this code scaffolded it with `hatch` so it probably makes sense to stick with it. I haven't worked with [...]

emmmmm kepler is based on golang... but hatch seems python? so maybe go fmt?

SamYuan1990, 2024-06-17:
> > > > what's `hatch fmt`?
> > > 
> > > 
> > > See: https://hatch.pypa.io/latest/cli/reference/#hatch-fmt
> > > Looks like whomever contributed this code scaffolded it with `hatch` so it probably makes sense to stick with it. I haven't worked with Python in a very long time so I have no opinions on whether this is right/wrong or best practice.
> > 
> > 
> > emmmmm kepler is based on golang... but hatch seems python? so maybe go fmt?
> 
> Correct. Except we seem to be collecting python code:
> 
> ```
> $ find . -iname "*.py"
> ./e2e/tools/validator/src/validator/cli/__init__.py
> ./e2e/tools/validator/src/validator/stresser/__init__.py
> ./e2e/tools/validator/src/validator/config/__init__.py
> ./e2e/tools/validator/src/validator/prometheus/__init__.py
> ./e2e/tools/validator/src/validator/__about__.py
> ./e2e/tools/validator/src/validator/__init__.py
> ./e2e/tools/validator/src/validator/__main__.py
> ./e2e/tools/validator/src/validator/cases/__init__.py
> ./e2e/tools/validator/src/validator/specs/__init__.py
> ./e2e/tools/validator/tests/validator/config/test_config.py
> ./e2e/tools/validator/tests/validator/prometheus/test_prometheus.py
> ./e2e/tools/validator/tests/validator/cases/test_cases.py
> ./e2e/tools/validator/tests/validator/stresser/test_stresser.py
> ./e2e/tools/validator/tests/__init__.py
> ./manifests/compose/mock-acpi/mock-acpi/mock-acpi.py
> ./manifests/compose/mock-acpi/turbostat/run_turbostat.py
> ```

understand. +1

vprashar2929, 2024-07-04:
closing as it is addressed in #1570 and #1591

closed 2024-07-04 by vprashar2929