sustainable-computing-io/kepler #1553: [ref] default bpf sample rate

jtaleric, 2024-06-18:
### What would you like to be added?

We should consider dropping the sampling rate to `EXPERIMENTAL_BPF_SAMPLE_RATE: 1000` 

We have seen decent improvement from the CPU utilization side when reducing the sampling rate. 

Risk - We have not quantified the possible loss of granularity of the power data when enabling this, however when comparing to our redfish raw data it is still very close. 
[image]

### Why is this needed?

CPU usage reduction.

dave-tucker, 2024-06-18:
We'll need to retest on `main` since #1481 has changed what's done before the sampling check in the eBPF code vs. what was done in the last released version. There might not be an appreciable difference in probe execution time with/without sampling set. I'll set up some micro-benchmarks to confirm once #1438 has gone in since that's blocking easier benchmarking/testing of the eBPF code.

dave-tucker, 2024-07-11:
Related to #1611 and #1607

closed 2026-05-07 by github-actions[bot]