sustainable-computing-io/kepler #1556: ARM support for AWS Graviton v2, v3

jjo, 2024-06-18:
### What would you like to be added?

ARM support for [AWS Graviton v2, v3](https://en.wikipedia.org/wiki/AWS_Graviton)

### Why is this needed?

Expanding Kepler support to more ARM architectures will be very beneficial, else we'd be subsetting our energy observability features to Intel-only, especially for commonly used architectures in some cloud providers: for AWS these are GravitonV2 for the `𝑥6g...`  (`c6g`, `m6g`, etc) and GravitonV3 for `𝑥7g...` (`c7g`, `m7g`, etc).

Worth noting previous issue at https://github.com/sustainable-computing-io/kepler/issues/482#issuecomment-1569995309.

I tried deploying `kepler-0.7.10` on an `m6g` instance with below details:
```
[root@ip-10-60-5-69 ~]# lscpu 
Architecture:        aarch64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
CPU(s):              32
On-line CPU(s) list: 0-31
Thread(s) per core:  1
Core(s) per socket:  32
Socket(s):           1
NUMA node(s):        1
Vendor ID:           ARM
Model:               1
Model name:          Neoverse-N1
Stepping:            r3p1
BogoMIPS:            243.75
L1d cache:           64K
L1i cache:           64K
L2 cache:            1024K
L3 cache:            32768K
NUMA node0 CPU(s):   0-31
Flags:               fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp ssbs
[root@ip-10-60-5-69 ~]# uname -r
5.10.215-203.850.amzn2.aarch64
```

It crashed with the below log tail (full log at https://0x0.st/XTMr.txt):

```
[...]
libbpf: failed to open '/sys/kernel/debug/tracing/events/writeback/writeback_dirty_folio/id': No such file or directory
libbpf: failed to determine tracepoint 'writeback/writeback_dirty_folio' perf event ID: No such file or directory
libbpf: prog 'kepler_write_page_trace': failed to create tracepoint 'writeback/writeback_dirty_folio' perf event: No such file or directory
W0618 20:16:47.934582       1 exporter.go:215] failed to attach tp/writeback/writeback_dirty_folio: failed to attach tracepoint writeback_dirty_folio to program kepler_write_page_trace: no such file or directory. Kepler will not collect page cache write events. This will affect the DRAM power model estimation on VMs.
libbpf: prog 'kepler_read_page_trace': failed to attach: ERROR: strerror_r(-524)=22
W0618 20:16:47.934648       1 exporter.go:227] failed to attach fentry/mark_page_accessed: failed to attach program: errno 524. Kepler will not collect page cache read events. This will affect the DRAM power model estimation on VMs.
I0618 20:16:48.149175       1 exporter.go:270] Successfully load eBPF module from libbpf object
I0618 20:16:48.149206       1 exporter.go:116] Initializing the GPU collector
I0618 20:16:48.149517       1 watcher.go:67] Using in cluster k8s config
I0618 20:16:48.650362       1 watcher.go:138] k8s APIserver watcher was started
I0618 20:16:48.650507       1 prometheus_collector.go:92] Registered Container Prometheus metrics
I0618 20:16:48.650548       1 prometheus_collector.go:97] Registered VM Prometheus metrics
I0618 20:16:48.650569       1 prometheus_collector.go:101] Registered Node Prometheus metrics
panic: runtime error: invalid memory address or nil pointer dereference
```

/cc @nikimanoledaki

dave-tucker, 2024-07-11:
@jjo if you still have the environment available could you test with Kepler v0.7.11?
I'm hopeful that this has fixed a couple of issues in ARM64 environments.

petewall, 2024-07-11:
I had replicated this with a graviton node. Let me see if I can do it again with 0.7.11

jjo, 2024-07-11:
> @jjo if you still have the environment available could you test with Kepler v0.7.11? I'm hopeful that this has fixed a couple of issues in ARM64 environments.

awesome, lemme check and report back

petewall, 2024-07-11:
It feels like it gets further, but still crashes:
```
libbpf: map 'cpu_freq_array': at sec_idx 13, offset 320.
libbpf: map 'cpu_freq_array': found type = 2.
libbpf: map 'cpu_freq_array': found key [6], sz = 4.
libbpf: map 'cpu_freq_array': found value [6], sz = 4.
libbpf: map 'cpu_freq_array': found max_entries = 128.
libbpf: map '.rodata.config' (global data): at sec_idx 11, offset 0, flags 80.
libbpf: map 11 is ".rodata.config"
libbpf: map 'kepler.bss' (global data): at sec_idx 12, offset 0, flags 400.
libbpf: map 12 is "kepler.bss"
libbpf: sec '.reltp/sched/sched_switch': collecting relocation for section(3) 'tp/sched/sched_switch'
libbpf: sec '.reltp/sched/sched_switch': relo #0: insn #0 against '.rodata.config'
libbpf: prog 'kepler_sched_switch_trace': found data map 11 (.rodata.config, sec 11, off 0) for insn 0
libbpf: sec '.reltp/sched/sched_switch': relo #1: insn #7 against 'counter_sched_switch'
libbpf: prog 'kepler_sched_switch_trace': found data map 12 (kepler.bss, sec 12, off 0) for insn 7
libbpf: sec '.reltp/sched/sched_switch': relo #2: insn #16 against '.rodata.config'
libbpf: prog 'kepler_sched_switch_trace': found data map 11 (.rodata.config, sec 11, off 0) for insn 16
libbpf: sec '.reltp/sched/sched_switch': relo #3: insn #41 against 'cpu_cycles_event_reader'
libbpf: prog 'kepler_sched_switch_trace': found map 2 (cpu_cycles_event_reader, sec 13, off 64) for insn #41
libbpf: sec '.reltp/sched/sched_switch': relo #4: insn #53 against 'cpu_cycles'
libbpf: prog 'kepler_sched_switch_trace': found map 3 (cpu_cycles, sec 13, off 96) for insn #53
libbpf: sec '.reltp/sched/sched_switch': relo #5: insn #67 against 'cpu_cycles'
libbpf: prog 'kepler_sched_switch_trace': found map 3 (cpu_cycles, sec 13, off 96) for insn #67
libbpf: sec '.reltp/sched/sched_switch': relo #6: insn #77 against 'cpu_instructions_event_reader'
libbpf: prog 'kepler_sched_switch_trace': found map 4 (cpu_instructions_event_reader, sec 13, off 128) for insn #77
libbpf: sec '.reltp/sched/sched_switch': relo #7: insn #87 against 'cpu_instructions'
libbpf: prog 'kepler_sched_switch_trace': found map 5 (cpu_instructions, sec 13, off 160) for insn #87
libbpf: sec '.reltp/sched/sched_switch': relo #8: insn #99 against 'cpu_instructions'
libbpf: prog 'kepler_sched_switch_trace': found map 5 (cpu_instructions, sec 13, off 160) for insn #99
libbpf: sec '.reltp/sched/sched_switch': relo #9: insn #111 against 'cache_miss_event_reader'
libbpf: prog 'kepler_sched_switch_trace': found map 6 (cache_miss_event_reader, sec 13, off 192) for insn #111
libbpf: sec '.reltp/sched/sched_switch': relo #10: insn #122 against 'cache_miss'
libbpf: prog 'kepler_sched_switch_trace': found map 7 (cache_miss, sec 13, off 224) for insn #122
libbpf: sec '.reltp/sched/sched_switch': relo #11: insn #136 against 'cache_miss'
libbpf: prog 'kepler_sched_switch_trace': found map 7 (cache_miss, sec 13, off 224) for insn #136
libbpf: sec '.reltp/sched/sched_switch': relo #12: insn #145 against 'pid_time'
libbpf: prog 'kepler_sched_switch_trace': found map 1 (pid_time, sec 13, off 32) for insn #145
libbpf: sec '.reltp/sched/sched_switch': relo #13: insn #153 against 'pid_time'
libbpf: prog 'kepler_sched_switch_trace': found map 1 (pid_time, sec 13, off 32) for insn #153
libbpf: sec '.reltp/sched/sched_switch': relo #14: insn #165 against 'pid_time'
libbpf: prog 'kepler_sched_switch_trace': found map 1 (pid_time, sec 13, off 32) for insn #165
libbpf: sec '.reltp/sched/sched_switch': relo #15: insn #171 against 'processes'
libbpf: prog 'kepler_sched_switch_trace': found map 0 (processes, sec 13, off 0) for insn #171
libbpf: sec '.reltp/sched/sched_switch': relo #16: insn #195 against 'processes'
libbpf: prog 'kepler_sched_switch_trace': found map 0 (processes, sec 13, off 0) for insn #195
libbpf: sec '.reltp/sched/sched_switch': relo #17: insn #227 against 'processes'
libbpf: prog 'kepler_sched_switch_trace': found map 0 (processes, sec 13, off 0) for insn #227
libbpf: sec '.reltp/irq/softirq_entry': collecting relocation for section(5) 'tp/irq/softirq_entry'
libbpf: sec '.reltp/irq/softirq_entry': relo #0: insn #6 against 'processes'
libbpf: prog 'kepler_irq_trace': found map 0 (processes, sec 13, off 0) for insn #6
libbpf: sec '.relfexit/mark_page_accessed': collecting relocation for section(7) 'fexit/mark_page_accessed'
libbpf: sec '.relfexit/mark_page_accessed': relo #0: insn #4 against 'processes'
libbpf: prog 'kepler_read_page_trace': found map 0 (processes, sec 13, off 0) for insn #4
libbpf: sec '.reltp/writeback/writeback_dirty_folio': collecting relocation for section(9) 'tp/writeback/writeback_dirty_folio'
libbpf: sec '.reltp/writeback/writeback_dirty_folio': relo #0: insn #4 against 'processes'
libbpf: prog 'kepler_write_page_trace': found map 0 (processes, sec 13, off 0) for insn #4
I0711 13:26:45.390256       1 exporter.go:158] 1 CPU cores detected. Resizing eBPF Perf Event Arrays
libbpf: loading kernel BTF '/sys/kernel/btf/vmlinux': 0
libbpf: map 'processes': created successfully, fd=8
libbpf: map 'pid_time': created successfully, fd=9
libbpf: map 'cpu_cycles_event_reader': created successfully, fd=10
libbpf: map 'cpu_cycles': created successfully, fd=11
libbpf: map 'cpu_instructions_event_reader': created successfully, fd=12
libbpf: map 'cpu_instructions': created successfully, fd=13
libbpf: map 'cache_miss_event_reader': created successfully, fd=14
libbpf: map 'cache_miss': created successfully, fd=15
[... cut ...]
    /workspace/pkg/metrics/container/metrics.go:102 +0xf4
github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
    /workspace/vendor/github.com/prometheus/client_golang/prometheus/registry.go:455 +0xd8
created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather in goroutine 1
    /workspace/vendor/github.com/prometheus/client_golang/prometheus/registry.go:547 +0x974
Stream closed EOF for default/kepler-8x9f8 (kepler-exporter)
```

This is using a `c6g.medium` machine type, so AWS Graviton2.

petewall, 2024-07-11:
Running again with debug logs...

petewall, 2024-07-11:
Logs again with log level 5:
```
I0711 13:36:58.394721       1 gpu.go:38] Trying to initialize GPU collector using dcgm
W0711 13:36:58.395340       1 gpu_dcgm.go:104] There is no DCGM daemon running in the host: libdcgm.so not Found
W0711 13:36:58.395447       1 gpu_dcgm.go:108] Could not start DCGM. Error: libdcgm.so not Found
I0711 13:36:58.395455       1 gpu.go:45] Error initializing dcgm: not able to connect to DCGM: libdcgm.so not Found
I0711 13:36:58.395460       1 gpu.go:38] Trying to initialize GPU collector using nvidia-nvml
I0711 13:36:58.395623       1 gpu.go:45] Error initializing nvidia-nvml: failed to init nvml. ERROR_LIBRARY_NOT_FOUND
I0711 13:36:58.395720       1 gpu.go:38] Trying to initialize GPU collector using dummy
I0711 13:36:58.395780       1 gpu.go:42] Using dummy to obtain gpu power
E0711 13:36:58.399807       1 utils.go:139] getCPUArch failure: open /sys/devices/cpu/caps/pmu_name: no such file or directory
I0711 13:36:58.399909       1 exporter.go:85] Kepler running on version: release-0.7.10
I0711 13:36:58.399921       1 config.go:283] using gCgroup ID in the BPF program: true
I0711 13:36:58.399944       1 config.go:285] kernel version: 5.1
I0711 13:36:58.399978       1 config.go:310] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0711 13:36:58.399987       1 config.go:151] ENABLE_EBPF_CGROUPID: false
I0711 13:36:58.399990       1 config.go:152] ENABLE_GPU: true
I0711 13:36:58.399994       1 config.go:153] ENABLE_QAT: false
I0711 13:36:58.399998       1 config.go:154] ENABLE_PROCESS_METRICS: false
I0711 13:36:58.400002       1 config.go:155] EXPOSE_HW_COUNTER_METRICS: true
I0711 13:36:58.400007       1 config.go:156] EXPOSE_CGROUP_METRICS: false
I0711 13:36:58.400011       1 config.go:157] EXPOSE_IRQ_COUNTER_METRICS: true
I0711 13:36:58.400016       1 config.go:158] EXPOSE_ESTIMATED_IDLE_POWER_METRICS: true. This only impacts when the power is estimated using pre-prained models. Estimated idle power is meaningful only when Kepler is running on bare-metal or with a single virtual machine (VM) on the node.
I0711 13:36:58.400023       1 config.go:159] EXPERIMENTAL_BPF_SAMPLE_RATE: 0
I0711 13:36:58.400051       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0711 13:36:58.400080       1 power.go:72] Unable to obtain power, use estimate method
I0711 13:36:58.400086       1 redfish.go:169] failed to get redfish credential file path
I0711 13:36:58.400686       1 acpi.go:71] Could not find any ACPI power meter path. Is it a VM?
I0711 13:36:58.400700       1 power.go:73] using none to obtain power
libbpf: loading /var/lib/kepler/bpfassets/kepler.bpfel.o
libbpf: elf: section(3) tp/sched/sched_switch, size 1864, link 0, flags 6, type=1
libbpf: sec 'tp/sched/sched_switch': found program 'kepler_sched_switch_trace' at insn offset 0 (0 bytes), code size 233 insns (1864 bytes)
libbpf: elf: section(4) .reltp/sched/sched_switch, size 288, link 32, flags 40, type=9
libbpf: elf: section(5) tp/irq/softirq_entry, size 144, link 0, flags 6, type=1
libbpf: sec 'tp/irq/softirq_entry': found program 'kepler_irq_trace' at insn offset 0 (0 bytes), code size 18 insns (144 bytes)
libbpf: elf: section(6) .reltp/irq/softirq_entry, size 16, link 32, flags 40, type=9
libbpf: elf: section(7) fexit/mark_page_accessed, size 104, link 0, flags 6, type=1
libbpf: sec 'fexit/mark_page_accessed': found program 'kepler_read_page_trace' at insn offset 0 (0 bytes), code size 13 insns (104 bytes)
libbpf: elf: section(8) .relfexit/mark_page_accessed, size 16, link 32, flags 40, type=9
libbpf: elf: section(9) tp/writeback/writeback_dirty_folio, size 104, link 0, flags 6, type=1
libbpf: sec 'tp/writeback/writeback_dirty_folio': found program 'kepler_write_page_trace' at insn offset 0 (0 bytes), code size 13 insns (104 bytes)
libbpf: elf: section(10) .reltp/writeback/writeback_dirty_folio, size 16, link 32, flags 40, type=9
libbpf: elf: section(11) .rodata.config, size 12, link 0, flags 2, type=1
libbpf: elf: section(12) .bss, size 4, link 0, flags 3, type=8
libbpf: elf: section(13) .maps, size 352, link 0, flags 3, type=1
libbpf: elf: section(14) license, size 13, link 0, flags 3, type=1
libbpf: license of /var/lib/kepler/bpfassets/kepler.bpfel.o is Dual BSD/GPL
libbpf: elf: section(23) .BTF, size 5659, link 0, flags 0, type=1
libbpf: elf: section(25) .BTF.ext, size 2072, link 0, flags 0, type=1
libbpf: elf: section(32) .symtab, size 1128, link 1, flags 0, type=2
libbpf: looking for externs among 47 symbols...
libbpf: collected 0 externs total
libbpf: map 'processes': at sec_idx 13, offset 0.
libbpf: map 'processes': found type = 1.
libbpf: map 'processes': found key [6], sz = 4.
libbpf: map 'processes': found value [10], sz = 112.
libbpf: map 'processes': found max_entries = 32768.
libbpf: map 'pid_time': at sec_idx 13, offset 32.
libbpf: map 'pid_time': found type = 1.
libbpf: map 'pid_time': found key [6], sz = 4.
libbpf: map 'pid_time': found value [12], sz = 8.
libbpf: map 'pid_time': found max_entries = 32768.
libbpf: map 'cpu_cycles_event_reader': at sec_idx 13, offset 64.
libbpf: map 'cpu_cycles_event_reader': found type = 4.
libbpf: map 'cpu_cycles_event_reader': found key [2], sz = 4.
libbpf: map 'cpu_cycles_event_reader': found value [6], sz = 4.
libbpf: map 'cpu_cycles_event_reader': found max_entries = 128.
libbpf: map 'cpu_cycles': at sec_idx 13, offset 96.
libbpf: map 'cpu_cycles': found type = 2.
[... cut ...]
github.com/sustainable-computing-io/kepler/pkg/metrics/container.(*collector).Collect(0x40003d4100, 0x4000069f58?)
	/workspace/pkg/metrics/container/metrics.go:102 +0xf4
github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
	/workspace/vendor/github.com/prometheus/client_golang/prometheus/registry.go:455 +0xd8
created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather in goroutine 1
	/workspace/vendor/github.com/prometheus/client_golang/prometheus/registry.go:547 +0x974
```

jjo, 2024-07-11:
> @jjo if you still have the environment available could you test with Kepler v0.7.11? I'm hopeful that this has fixed a couple of issues in ARM64 environments.

woOT!, running ok so far:

```
I0711 13:32:10.264592       1 gpu.go:38] Trying to initialize GPU collector using dcgm
W0711 13:32:10.264788       1 gpu_dcgm.go:104] There is no DCGM daemon running in the host: libdcgm.so not Found
W0711 13:32:10.264815       1 gpu_dcgm.go:108] Could not start DCGM. Error: libdcgm.so not Found
I0711 13:32:10.264826       1 gpu.go:45] Error initializing dcgm: not able to connect to DCGM: libdcgm.so not Found
I0711 13:32:10.264830       1 gpu.go:38] Trying to initialize GPU collector using nvidia-nvml
I0711 13:32:10.264880       1 gpu.go:45] Error initializing nvidia-nvml: failed to init nvml. ERROR_LIBRARY_NOT_FOUND
I0711 13:32:10.264889       1 gpu.go:38] Trying to initialize GPU collector using dummy
I0711 13:32:10.264893       1 gpu.go:42] Using dummy to obtain gpu power
E0711 13:32:10.265472       1 utils.go:110] getCPUArch failure: open /sys/devices/cpu/caps/pmu_name: no such file or directory
I0711 13:32:10.266033       1 exporter.go:100] Kepler running on version: v0.7.11
I0711 13:32:10.266089       1 config.go:284] using gCgroup ID in the BPF program: true
I0711 13:32:10.266129       1 config.go:286] kernel version: 5.1
I0711 13:32:10.266162       1 config.go:311] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0711 13:32:10.266199       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0711 13:32:10.266230       1 power.go:72] Unable to obtain power, use estimate method
I0711 13:32:10.266242       1 redfish.go:169] failed to get redfish credential file path
I0711 13:32:10.266638       1 acpi.go:71] Could not find any ACPI power meter path. Is it a VM?
I0711 13:32:10.266652       1 power.go:73] using none to obtain power
I0711 13:32:10.364233       1 exporter.go:89] Number of CPUs: 32
W0711 13:32:15.972242       1 exporter.go:150] failed to attach fentry/mark_page_accessed: create raw tracepoint: not supported. Kepler will not collect page cache read events. This will affect the DRAM power model estimation on VMs.
I0711 13:32:16.296235       1 exporter.go:147] Initializing the GPU collector
I0711 13:32:16.296566       1 watcher.go:68] Using in cluster k8s config
I0711 13:32:16.862345       1 watcher.go:140] k8s APIserver watcher was started
I0711 13:32:16.862410       1 prometheus_collector.go:95] Registered Container Prometheus metrics
I0711 13:32:16.862451       1 prometheus_collector.go:100] Registered VM Prometheus metrics
I0711 13:32:16.862476       1 prometheus_collector.go:104] Registered Node Prometheus metrics
I0711 13:32:16.965248       1 process_energy.go:114] Using the Ratio/DynPower Power Model to estimate Process Platform Power
I0711 13:32:16.965271       1 process_energy.go:115] Process feature names: [bpf_cpu_time_ms]
I0711 13:32:16.965294       1 process_energy.go:124] Using the Ratio/DynPower Power Model to estimate Process Component Power
I0711 13:32:16.965301       1 process_energy.go:125] Process feature names: [bpf_cpu_time_ms bpf_cpu_time_ms bpf_cpu_time_ms   gpu_compute_util]
I0711 13:32:16.965466       1 node_platform_energy.go:52] Using the Regressor/AbsPower Power Model to estimate Node Platform Power
I0711 13:32:16.965540       1 node_component_energy.go:56] Using the Regressor/AbsPower Power Model to estimate Node Component Power
I0711 13:32:16.965581       1 exporter.go:201] starting to listen on 0.0.0.0:9102
I0711 13:32:16.965696       1 exporter.go:215] Started Kepler in 6.699690316s
```

More details re:running instance:

```
# uname -a
Linux ip-10-60-8-147.us-east-2.compute.internal 5.10.219-208.866.amzn2.aarch64 #1 SMP Tue Jun 18 14:00:02 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

# lscpu 
Architecture:        aarch64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
CPU(s):              32
On-line CPU(s) list: 0-31
Thread(s) per core:  1
Core(s) per socket:  32
Socket(s):           1
NUMA node(s):        1
Vendor ID:           ARM
Model:               1
Model name:          Neoverse-N1
Stepping:            r3p1
BogoMIPS:            243.75
L1d cache:           64K
L1i cache:           64K
L2 cache:            1024K
L3 cache:            32768K
NUMA node0 CPU(s):   0-31
Flags:               fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp ssbs
```
```

petewall, 2024-07-11:
Ahh geezz... Just realized I'm still running 0.7.10.. Sorry for the noise. Let me update and I'll let you know

dave-tucker, 2024-07-11:
> Ahh geezz... Just realized I'm still running 0.7.10.. Sorry for the noise. Let me update and I'll let you know

No worries I was about to say "those look like 0.7.10 logs" 🤣 
Hoping this at least doesn't crash this time around 😓

petewall, 2024-07-11:
> > Ahh geezz... Just realized I'm still running 0.7.10.. Sorry for the noise. Let me update and I'll let you know
> 
> No worries I was about to say "those look like 0.7.10 logs" 🤣 Hoping this at least doesn't crash this time around 😓

Sorry for the short term panic!

Good news! 0.7.11 is no longer crashing on my graviton node, too!

dave-tucker, 2024-07-11:
Given that things aren't crashing I'm going to go ahead and close this one for now.

Maarham, 2024-07-12:
Hey, just wanted to follow up on this thread. When you are saying no crashes, are y'all able to get information (valid sustainability metrics like power consumption, carbon emissions, etc.) from Kepler? Do they seem valid and how do the results look like? Thanks!

nikimanoledaki, 2026-11-14:
Reopening since this was closed when Kepler stopped crashing, but [the question](https://github.com/sustainable-computing-io/kepler/issues/1556#issuecomment-2226346143) about whether the metrics are valid was not answered.

As of v0.12.0, Kepler builds for arm64 (#2442) and falls back to HWMON when RAPL isn't available (#2440, #2428). What's still unknown is whether Graviton instances expose any HWMON power or energy sensors at all. If they don't, Kepler will deploy but have nothing real to measure.

To check on a Graviton node (e.g. `m6g`/`m7g`):

```sh
for d in /sys/class/hwmon/hwmon*; do
  echo "== $d: $(cat $d/name)"
  ls $d | grep -E '^(energy|power|curr|in)[0-9]+_(input|label)$'
done
```

If any `energy*`/`power*`/`curr*` files show up, please deploy v0.12.0 and share the logs and `kepler_node_cpu_watts`. 

Until #2638 lands, the published `v0.12.0` image is amd64 only, so build it with `make image GOARCH=arm64`. If nothing shows up, that's useful to know too: we'd then need another source for Graviton, such as a model-based estimate.