sustainable-computing-io/kepler #1607: bpf sampling no longer reduces overhead

rootfs, 2024-07-09:
### What happened?

Latest the `EXPERIMENTAL_BPF_SAMPLE_RATE` in kepler ebpf code no longer shows any difference in reducing ebpf overhead.

I compared 0.7.8 and latest code:
[image]

### What did you expect to happen?

EXPERIMENTAL_BPF_SAMPLE_RATE should be able to reduce ebpf overhead 

### How can we reproduce it (as minimally and precisely as possible)?

Run the bpftool command on kepler latest and 0.7.8

### Kepler image tag

<details>
0.7.8 and latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>
KVM 
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE}
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE}
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2024-07-09:
@sthaha @dave-tucker can you get this fixed before 0.7.11 release? thanks

sthaha, 2024-07-10:
@rootfs I ran the same on my machine and here are the results 

```
# no sampling
❯ sudo bpftool prog show name kepler_sched_switch_trace | head -n 1 | awk '{print "rt_ns:", $(NF-2), "count: ", $NF, "avg: ", $(NF-2)/$NF } ' | tee -a kepler-0.log
rt_ns: 572554527 count:  129051 avg:  4436.65

```

```
# sample rate: 1000
❯ sudo bpftool prog show name kepler_sched_switch_trace | head -n 1 | awk '{print "rt_ns:", $(NF-2), "count: ", $NF, "avg: ", $(NF-2)/$NF } ' | tee -a kepler-0.log
rt_ns: 1355770684 count:  481430 avg:  2816.13
```

I see that this happens because sampling is not  honoured (on purpose) in https://github.com/sustainable-computing-io/kepler/blob/d3c2906c9abd0052d38e75c19e1577dead497e5c/bpf/kepler.bpf.c#L210-L213

```c

	// Collect metrics
	// Regardless of skipping the collection, we need to update the hardware
	// counter events to keep the metrics map current.
	collect_metrics_and_reset_counters(&buf, prev_pid, curr_ts, cpu_id);

```

@marceloamaral could you please  confirm this is necessary?

vimalk78, 2024-07-11:
in the current way of sampling we sample the hardware counters each time but we use it only when sampling period ends (counter goes to 0), and discard all the samples which is adding to the cost.

if we use the sample only when the counter goes to zero, why not skip sampling in the whole period except for when counter is 1 (which will also be discarded because the delta would be wrong)

trying a patch with this change, and testing

sthaha, 2024-07-11:
FYI: I have removed this from the 0.7.11 milestone. Lets address this issue in the 0.7.12 milestone.

dave-tucker, 2024-07-11:
Per @marceloamaral's comments I think correctness is important, and therefore sampling is no longer a viable approach to reduce the overhead we're adding on every `sched_switch` event.

I've put together an alternative proposal here: https://github.com/sustainable-computing-io/kepler/issues/1611 which details how we could move some of this logic to userland, which would achieve both objectives:

1. Correctness
2. Lower probe overhead

PR #1685 (2024-08-14): fix(bpf): Fix overhead when sampling
  Update the register counters one step before metrics sample is taken
  instead of updating registers every time which is increasing overhead

- with `EXPERIMENTAL_BPF_SAMPLE_RATE = 0`
```
[root@vimalkum-thinkpadp1gen4i ~]# bpftool prog show name kepler_sched_switch_trace | head -n 1 | awk '{print "rt_ns:", $(NF-2), "count: ", $NF, "avg: ", $(NF-2)/$NF } '
rt_ns: 301913885 count:  95144 avg:  3173.23

[root@vimalkum-thinkpadp1gen4i ~]# bpftool prog show name kepler_sched_switch_trace | head -n 1 | awk '{print "rt_ns:", $(NF-2), "count: ", $NF, "avg: ", $(N [... cut]
closed 2024-08-14 by rootfs