sustainable-computing-io/kepler #1640: Kepler latest reporting wrong command in process metrics

vprashar2929, 2024-07-25:
### What happened?

When Kepler using `latest` is deployed it reports the wrong command in the `kepler_process` metrics. 
Attaching screenshot for a comparison between what Kepler's latest and Scaphendre reports:

[image]

Here pid=`503764` which is actually pid for qemu process running on the machine
```
❯ pstree -p | grep 503764
           |-qemu-system-x86(503764)-+-{qemu-system-x86}(503771)
```

### What did you expect to happen?

Kepler should report the command name correctly in the metrics

### How can we reproduce it (as minimally and precisely as possible)?

Deploy Kepler latest using Docker compose and then check for commands in the Kepler process metrics for one of the pid

### Kepler image tag

<details>
latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE}
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE}
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

vprashar2929, 2024-07-31:
Verified locally and now Kepler latest is producing the correct command name for a particular pid

[image]

Thanks @dave-tucker for the fix!

vimalk78, 2024-08-14:
fixed by #1693

closed 2024-08-14 by vimalk78