sustainable-computing-io/kepler #1660: bpf: Investigate the *best* value for wakeup_data_size

dave-tucker, 2024-08-01:
### What would you like to be added?

This constant:
https://github.com/sustainable-computing-io/kepler/blob/main/bpf/kepler.bpf.c#L70

Declares how often we wake up to read the ringbuf.

The current math was as follows:

1. My system (on average) processes around 600-700 context switches per second
1. The sample period in Kepler is once every 3 seconds
1. We need to read at least one batch of ringbuf events within that 3 second interval

So that should have me read every 1.7ish seconds 😄 

### Why is this needed?

When kepler wakes up to read events it consumes CPU. Right now that's between 1-3%.
We should consider whether there is a better formula we could use to compute this magic number of 1000.

It could relate to the sample rate.

e.g `500 * SampleRate` and perhaps even the `500` could come from something better than an educated guess.

rootfs, 2024-08-01:
The Kepler CPU usage under normal and stress workloads need to be investigated in parallel. The latest stress test results point to a divergence that needs to be fixed.

rootfs, 2024-08-02:
The current Kepler CPU usage is now 20% without running load. 
[image]

rootfs, 2024-08-02:
[Test results](https://github.com/sustainable-computing-io/kepler/pull/1628#issuecomment-2266174953) posted on the original PR https://github.com/sustainable-computing-io/kepler/pull/1628

dave-tucker, 2024-08-05:
> The current Kepler CPU usage is now 20% without running load. [image]

How, and on what machine, can I reproduce this result?

rootfs, 2024-08-05:
@dave-tucker load the kepler latest image and keep it running for a day.

dave-tucker, 2024-08-05:
> [Test results](https://github.com/sustainable-computing-io/kepler/pull/1628#issuecomment-2266174953) posted on the original PR #1628

Responded:
https://github.com/sustainable-computing-io/kepler/pull/1628#issuecomment-2269058775