sustainable-computing-io/kepler #1669: integration_test / integration_test_with_libbpf blocks PR

sthaha, 2024-08-05:
E2E test has been failing more often that not unrelated to the change made in PR. 
See: https://github.com/sustainable-computing-io/kepler/actions/runs/10249144658/job/28352058144?pr=1668

sthaha, 2024-08-05:
I have assigned this to myself but please feel free to investigate if you have the bandwidth.

vprashar2929, 2024-08-05:
Something is wrong with Kepler latest on VM. Upon digging I figured out that commit after 287496a340a058c03c080b1582aa33e7d602773b might have caused this(Probably b424607400784bc1971d872856d09ef6c7650ace). Till this commit the e2e's are passing

rootfs, 2024-08-06:
i restarted failed jobs and some of them are green again.

rootfs, 2024-08-06:
the latest 2 is green while latest 1 failed https://github.com/sustainable-computing-io/kepler/actions/runs/10264137816/job/28434285849?pr=1676

rootfs, 2024-08-07:
I dumped the metrics before and after the e2e test here https://github.com/sustainable-computing-io/kepler/actions/runs/10276105835/job/28436054646?pr=1679

Before the e2e, `kepler_container_joules_total` is zero, but none zero after e2e. It could be that the metrics test hit the very first initial state when all metrics are zero. We should let the e2e to wait till the metrics are none zero (i.e. kepler actually starts working) @sthaha @vprashar2929 wdyt?

maryamtahhan, 2024-08-07:
So, I see something funny in the kind cluster, I have the following pods: 
[image]
When I query the pods on kind-worker I see the stats for the kepler-exporter pod itself as expected:
[image]
When I query the second kind worker for the kepler-exporter pod I don't see any
[image]
I went down this track after noticing that when I queried the deamonset metrics I could only see one set of stats in the kind cluster
[image]

vprashar2929, 2024-08-07:
I have also seen this locally. Kepler does produce metrics with a slight delay.

maryamtahhan, 2024-08-12:
I implemented something in this PR https://github.com/sustainable-computing-io/kepler/pull/1686 to help with requeuing pod events in the case where a container in a pod is marked as not ready so that it can be processed again and eventually the stats for it are created.

sunya-ch, 2024-08-13:
For some reason, Kepler-exporter pod metric is timely not captured even if it is running (as we can call /metrics API from it).
I tested a change to validate over the other common pod like `kube-apiserver-kind-control-plane` to confirm my PR valid for integration test. 

One choice to unblock the PR issue for now is to use one of previously-running control plane pod like kube-apiserver that we know for sure that it will produce some metrics. At the same time, we need to investigate the reason of missing metrics of the Kepler-exporter.

My temporary commit:
https://github.com/sustainable-computing-io/kepler/pull/1684/commits/f97a193b42ef6544b1daeda6aee8cd9cd6c039a4

I can create another clean PR for this task (change target from exporter pod to apiserver).

vprashar2929, 2024-08-13:
After reverting [this](https://github.com/sustainable-computing-io/kepler/pull/1692) the e2e passes. Trigger multiple CI runs all pass https://github.com/sustainable-computing-io/kepler/actions/runs/10368134089/job/28702642716?pr=1692

closed 2024-08-15 by rootfs

sunya-ch, 2024-08-15:
@rootfs  The PR hasn't merged yet. Shall we keep this open until we decide the merge the #1692?