sustainable-computing-io/kepler #1765: Error Loading eBPF Objects (softirq_entry)

marvin-steinke, 2024-09-05:
### What happened?

Kepler seems to have problems with eBPF on my current setup. Kepler logs state:

```
failed to create eBPF exporter: error loading eBPF objects: field KeplerIrqTrace: program kepler_irq_trace: attach Tracing/TraceRawTp: raw_tp softirq_entry not supported
```

However `softirq_entry` is present at `/sys/kernel/debug/` on the host. I did find the similar issue #2201 which points to to a permission problem. Do I need to configure my host differently?

### What did you expect to happen?

Installation succeeds.

### How can we reproduce it (as minimally and precisely as possible)?

`helm install kepler kepler/kepler --namespace kepler --create-namespace`

### Anything else we need to know?

OS: Ubuntu 20.04.3 LTS x86_64 
Host: SYS-1019GP-TT 0123456789 
Kernel: 5.4.0-192-generic 
CPU: Intel Xeon Silver 4208 (16) @ 3.200GHz 
GPU: NVIDIA Quadro RTX 5000 
GPU: NVIDIA Quadro RTX 5000 
Memory: 95208MiB

### Kepler image tag

<details>

I0905 08:13:52.281482       1 gpu.go:38] Trying to initialize GPU collector using dcgm
W0905 08:13:52.281702       1 gpu_dcgm.go:104] There is no DCGM daemon running in the host: libdcgm.so not Found
W0905 08:13:52.281727       1 gpu_dcgm.go:108] Could not start DCGM. Error: libdcgm.so not Found
I0905 08:13:52.281733       1 gpu.go:45] Error initializing dcgm: not able to connect to DCGM: libdcgm.so not Found
I0905 08:13:52.281739       1 gpu.go:38] Trying to initialize GPU collector using nvidia-nvml
I0905 08:13:52.281789       1 gpu.go:45] Error initializing nvidia-nvml: failed to init nvml. ERROR_LIBRARY_NOT_FOUND
I0905 08:13:52.281798       1 gpu.go:38] Trying to initialize GPU collector using dummy
I0905 08:13:52.281803       1 gpu.go:42] Using dummy to obtain gpu power
I0905 08:13:52.285110       1 exporter.go:100] Kepler running on version: v0.7.11
I0905 08:13:52.285158       1 config.go:284] using gCgroup ID in the BPF program: true
I0905 08:13:52.285182       1 config.go:286] kernel version: 5.4
I0905 08:13:52.285247       1 config.go:311] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0905 08:13:52.285302       1 power.go:53] use sysfs to obtain power
I0905 08:13:52.285315       1 redfish.go:169] failed to get redfish credential file path
I0905 08:13:52.289657       1 power.go:73] using acpi to obtain power
I0905 08:13:52.292851       1 exporter.go:89] Number of CPUs: 16
F0905 08:13:52.412014       1 exporter.go:140] failed to create eBPF exporter: error loading eBPF objects: field KeplerIrqTrace: program kepler_irq_trace: attach Tracing/TraceRawTp: raw_tp softirq_entry not supported

</details>

### Kubernetes version

<details>

```console
$ kubectl version
Client Version: v1.30.4+k3s1
Kustomize Version: v5.0.4-0.20230601165947-6ce0bf390ce3
Server Version: v1.30.4+k3s1

```

</details>

### Cloud provider or bare metal

Bare Meal

[... 131 lines cut ...]
                             node.kubernetes.io/memory-pressure:NoSchedule op=Exists
                             node.kubernetes.io/network-unavailable:NoSchedule op=Exists
                             node.kubernetes.io/not-ready:NoExecute op=Exists
                             node.kubernetes.io/pid-pressure:NoSchedule op=Exists
                             node.kubernetes.io/unreachable:NoExecute op=Exists
                             node.kubernetes.io/unschedulable:NoSchedule op=Exists
Events:
  Type     Reason     Age                 From               Message
  ----     ------     ----                ----               -------
  Normal   Scheduled  107s                default-scheduler  Successfully assigned kepler/kepler-9qknx to gpu01
  Normal   Pulled     107s                kubelet            Successfully pulled image "quay.io/sustainable_computing_io/kepler:release-0.7.11" in 564ms (564ms including waiting). Image size: 117793827 bytes.
  Normal   Pulled     106s                kubelet            Successfully pulled image "quay.io/sustainable_computing_io/kepler:release-0.7.11" in 459ms (459ms including waiting). Image size: 117793827 bytes.
  Normal   Pulled     90s                 kubelet            Successfully pulled image "quay.io/sustainable_computing_io/kepler:release-0.7.11" in 552ms (552ms including waiting). Image size: 117793827 bytes.
  Normal   Created    58s (x4 over 107s)  kubelet            Created container kepler-exporter
  Normal   Started    58s (x4 over 107s)  kubelet            Started container kepler-exporter
  Normal   Pulled     58s                 kubelet            Successfully pulled image "quay.io/sustainable_computing_io/kepler:release-0.7.11" in 490ms (490ms including waiting). Image size: 117793827 bytes.
  Warning  BackOff    23s (x8 over 105s)  kubelet            Back-off restarting failed container kepler-exporter in pod kepler-9qknx_kepler(63598b16-bf4d-4d1b-af97-55672ac817b4)
  Normal   Pulling    11s (x5 over 107s)  kubelet            Pulling image "quay.io/sustainable_computing_io/kepler:release-0.7.11"
```

</details>

### Container runtime (CRI) and version (if applicable)

Containerd v1.7.20-k3s1

marvin-steinke, 2024-09-13:
So I installed a newer version of the kernel and this fixed the issue. I think the minimum kernel requirements in the docs should be updated (or maybe I overlooked something?). I'd be happy to do this. Where do you think this should be stated best and what version is the minimum based on the eBPF features used?

dave-tucker, 2024-09-13:
Relates to: #1483

5.12 is the minimum supported kernel version: https://github.com/sustainable-computing-io/kepler/issues/1483#issuecomment-2144881310

closed 2024-09-15 by marvin-steinke