sustainable-computing-io/kepler #1783: GPU Collector Does Not Use DCGM nor NVML

marvin-steinke, 2024-09-13:
### What happened?

I have installed NVIDIA's gpu-operator to make gpus available in my cluster with the latest drivers. nvidia-nvml is installed on the system and the DCGM daemon ist also installed (even though there is already an exporter running through the gpu-operator helm chart). Kepler however, does not recognize any GPUs and uses the dummy instead.

<details>

<summary>logs</summary>

```console

I0913 13:39:15.572574       1 gpu.go:38] Trying to initialize GPU collector using dcgm
W0913 13:39:15.572838       1 gpu_dcgm.go:104] There is no DCGM daemon running in the host: libdcgm.so not Found
W0913 13:39:15.572877       1 gpu_dcgm.go:108] Could not start DCGM. Error: libdcgm.so not Found
I0913 13:39:15.572886       1 gpu.go:45] Error initializing dcgm: not able to connect to DCGM: libdcgm.so not Found
I0913 13:39:15.572893       1 gpu.go:38] Trying to initialize GPU collector using nvidia-nvml
I0913 13:39:15.572952       1 gpu.go:45] Error initializing nvidia-nvml: failed to init nvml. ERROR_LIBRARY_NOT_FOUND
I0913 13:39:15.572958       1 gpu.go:38] Trying to initialize GPU collector using dummy
I0913 13:39:15.572962       1 gpu.go:42] Using dummy to obtain gpu power
I0913 13:39:15.575216       1 exporter.go:100] Kepler running on version: v0.7.11
I0913 13:39:15.575269       1 config.go:284] using gCgroup ID in the BPF program: true
I0913 13:39:15.575315       1 config.go:286] kernel version: 6.8
I0913 13:39:15.575399       1 config.go:311] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0913 13:39:15.575443       1 power.go:53] use sysfs to obtain power
I0913 13:39:15.575452       1 redfish.go:169] failed to get redfish credential file path
I0913 13:39:15.580022       1 power.go:73] using acpi to obtain power
I0913 13:39:15.582149       1 exporter.go:89] Number of CPUs: 16
I0913 13:39:16.311226       1 exporter.go:147] Initializing the GPU collector
I0913 13:39:16.312205       1 watcher.go:68] Using in cluster k8s config
I0913 13:39:16.413694       1 watcher.go:140] k8s APIserver watcher was started
I0913 13:39:16.413830       1 prometheus_collector.go:95] Registered Container Prometheus metrics
I0913 13:39:16.413939       1 prometheus_collector.go:100] Registered VM Prometheus metrics
I0913 13:39:16.414001       1 prometheus_collector.go:104] Registered Node Prometheus metrics
I0913 13:39:16.417126       1 process_energy.go:114] Using the Ratio/DynPower Power Model to estimate Process Platform Power
I0913 13:39:16.417167       1 process_energy.go:115] Process feature names: [bpf_cpu_time_ms]
I0913 13:39:16.417235       1 process_energy.go:124] Using the Ratio/DynPower Power Model to estimate Process Component Power
I0913 13:39:16.417256       1 process_energy.go:125] Process feature names: [bpf_cpu_time_ms bpf_cpu_time_ms bpf_cpu_time_ms   gpu_compute_util]
I0913 13:39:16.417534       1 exporter.go:201] starting to listen on 0.0.0.0:9102
I0913 13:39:16.417732       1 exporter.go:215] Started Kepler in 842.551051ms

```

</details>

<details>

<summary>nvidia-dcgm.service</summary>

```console

● nvidia-dcgm.service - NVIDIA DCGM service
     Loaded: loaded (/usr/lib/systemd/system/nvidia-dcgm.service; enabled; preset: enabled)
     Active: active (running) since Fri 2024-09-13 11:10:00 UTC; 2h 26min ago
   Main PID: 1051 (nv-hostengine)
      Tasks: 8 (limit: 114158)
     Memory: 20.5M (peak: 21.2M)
        CPU: 1min 7.237s
     CGroup: /system.slice/nvidia-dcgm.service
             └─1051 /usr/bin/nv-hostengine -n --service-account nvidia-dcgm

Sep 13 11:10:00 gpu01 systemd[1]: Started nvidia-dcgm.service - NVIDIA DCGM service.
Sep 13 11:10:01 gpu01 nv-hostengine[1051]: DCGM initialized
Sep 13 11:10:01 gpu01 nv-hostengine[1051]: Started host engine version 3.3.8 using port number: 5555

```

</details>

<details>

<summary>nvidia-smi</summary>
[... 193 lines cut ...]
Node-Selectors:              kubernetes.io/os=linux
Tolerations:                 node-role.kubernetes.io/control-plane:NoSchedule
                             node.kubernetes.io/disk-pressure:NoSchedule op=Exists
                             node.kubernetes.io/memory-pressure:NoSchedule op=Exists
                             node.kubernetes.io/network-unavailable:NoSchedule op=Exists
                             node.kubernetes.io/not-ready:NoExecute op=Exists
                             node.kubernetes.io/pid-pressure:NoSchedule op=Exists
                             node.kubernetes.io/unreachable:NoExecute op=Exists
                             node.kubernetes.io/unschedulable:NoSchedule op=Exists
Events:
  Type    Reason     Age   From               Message
  ----    ------     ----  ----               -------
  Normal  Scheduled  16m   default-scheduler  Successfully assigned energy-aware/kepler-cw9sg to gpu01
  Normal  Pulling    16m   kubelet            Pulling image "quay.io/sustainable_computing_io/kepler:release-0.7.11"
  Normal  Pulled     16m   kubelet            Successfully pulled image "quay.io/sustainable_computing_io/kepler:release-0.7.11" in 488ms (488ms including waiting). Image size: 117793827 bytes.
  Normal  Created    16m   kubelet            Created container kepler-exporter
  Normal  Started    16m   kubelet            Started container kepler-exporter

</details>

### Container runtime (CRI) and version (if applicable)

<details>
Containerd v1.7.20-k3s1
</details>

marvin-steinke, 2024-09-16:
When I manually add `/usr/lib` (where nvidia-nvml is located) as a volume from the host to kepler's daemon set, kepler (or rather the go bindings library) find the library and uses it to obtain gpu power.

<details>
<summary>logs</summary>

```
I0916 12:47:19.740990       1 gpu.go:38] Trying to initialize GPU collector using dcgm
W0916 12:47:19.741684       1 gpu_dcgm.go:104] There is no DCGM daemon running in the host: libdcgm.so not Found
W0916 12:47:19.742091       1 gpu_dcgm.go:108] Could not start DCGM. Error: libdcgm.so not Found
I0916 12:47:19.742106       1 gpu.go:45] Error initializing dcgm: not able to connect to DCGM: libdcgm.so not Found
I0916 12:47:19.742113       1 gpu.go:38] Trying to initialize GPU collector using nvidia-nvml
I0916 12:47:19.751727       1 gpu.go:42] Using nvidia-nvml to obtain gpu power
I0916 12:47:19.754465       1 exporter.go:100] Kepler running on version: v0.7.11
I0916 12:47:19.754522       1 config.go:284] using gCgroup ID in the BPF program: true
I0916 12:47:19.754576       1 config.go:286] kernel version: 6.8
I0916 12:47:19.754691       1 config.go:311] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0916 12:47:19.754750       1 power.go:53] use sysfs to obtain power
I0916 12:47:19.754765       1 redfish.go:169] failed to get redfish credential file path
I0916 12:47:19.759634       1 power.go:73] using acpi to obtain power
I0916 12:47:19.762127       1 exporter.go:89] Number of CPUs: 16
I0916 12:47:20.384788       1 exporter.go:147] Initializing the GPU collector
I0916 12:47:20.384850       1 gpu_nvml.go:79] found 2 gpu devices
I0916 12:47:20.390448       1 gpu_nvml.go:91] GPU 0 "Quadro RTX 5000" "GPU-de213ed3-8842-6af0-c3d6-3ad4669aac5a"
I0916 12:47:20.396092       1 gpu_nvml.go:91] GPU 1 "Quadro RTX 5000" "GPU-b5b7ae74-8a9b-85ff-7709-12c0cb2b4c7a"
I0916 12:47:20.397173       1 watcher.go:68] Using in cluster k8s config
I0916 12:47:20.497763       1 watcher.go:140] k8s APIserver watcher was started
I0916 12:47:20.497885       1 prometheus_collector.go:95] Registered Container Prometheus metrics
I0916 12:47:20.497953       1 prometheus_collector.go:100] Registered VM Prometheus metrics
I0916 12:47:20.497988       1 prometheus_collector.go:104] Registered Node Prometheus metrics
I0916 12:47:20.501595       1 process_energy.go:114] Using the Ratio/DynPower Power Model to estimate Process Platform Power
I0916 12:47:20.501610       1 process_energy.go:115] Process feature names: [bpf_cpu_time_ms]
I0916 12:47:20.501635       1 process_energy.go:124] Using the Ratio/DynPower Power Model to estimate Process Component Power
I0916 12:47:20.501643       1 process_energy.go:125] Process feature names: [bpf_cpu_time_ms bpf_cpu_time_ms bpf_cpu_time_ms   gpu_compute_util]
I0916 12:47:20.501807       1 exporter.go:201] starting to listen on 0.0.0.0:9102
I0916 12:47:20.501977       1 exporter.go:215] Started Kepler in 747.532396ms
```

</details>

I don't think this is the intended way.  Did I install the driver or the container toolkit wrong?

```console
$ sudo find / -name libnvidia-ml.so
/usr/lib/x86_64-linux-gnu/libnvidia-ml.so
/usr/lib/x86_64-linux-gnu/stubs/libnvidia-ml.so
/var/lib/rancher/k3s/agent/containerd/io.containerd.snapshotter.v1.overlayfs/snapshots/1000/fs/usr/local/cuda-11.8/targets/x86_64-linux/lib/stubs/libnvidia-ml.so
/var/lib/rancher/k3s/agent/containerd/io.containerd.snapshotter.v1.overlayfs/snapshots/788/fs/usr/local/cuda-11.8/targets/x86_64-linux/lib/stubs/libnvidia-ml.so
/var/lib/rancher/k3s/agent/containerd/io.containerd.snapshotter.v1.overlayfs/snapshots/682/fs/usr/local/cuda-11.8/targets/x86_64-linux/lib/stubs/libnvidia-ml.so
```

maryamtahhan, 2024-09-16:
It looks like it's using libnvml:

```
I0916 12:47:19.742113       1 gpu.go:38] Trying to initialize GPU collector using nvidia-nvml
I0916 12:47:19.751727       1 gpu.go:42] Using nvidia-nvml to obtain gpu power
```

and 
```
I0916 12:47:20.390448       1 gpu_nvml.go:91] GPU 0 "Quadro RTX 5000" "GPU-de213ed3-8842-6af0-c3d6-3ad4669aac5a"
I0916 12:47:20.396092       1 gpu_nvml.go:91] GPU 1 "Quadro RTX 5000" "GPU-b5b7ae74-8a9b-85ff-7709-12c0cb2b4c7a"
```

maryamtahhan, 2024-09-16:
There are different Kepler images: https://quay.io/repository/sustainable_computing_io/kepler?tab=tags
I think if you want to use the image that supports Nvidia GPUs you would need to use the `latest-dcgm` image rather than the `latest` image

marvin-steinke, 2024-09-16:
> There are different Kepler images: https://quay.io/repository/sustainable_computing_io/kepler?tab=tags I think if you want to use the image that supports Nvidia GPUs you would need to use the `latest-dcgm` image rather than the `latest` image

I tried that before but it didn't help:

<details>
<summary>kepler logs with "latest-dcgm" tag</summary>

```
I0916 13:14:44.689746       1 exporter.go:103] Kepler running on version: v0.7.11-227-g8a3587aa-dirty
I0916 13:14:44.689863       1 config.go:293] using gCgroup ID in the BPF program: true
I0916 13:14:44.689882       1 config.go:295] kernel version: 6.8
I0916 13:14:44.689995       1 config.go:322] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0916 13:14:44.690073       1 power.go:59] use sysfs to obtain power
I0916 13:14:44.690088       1 redfish.go:169] failed to get redfish credential file path
I0916 13:14:44.694713       1 power.go:79] using acpi to obtain power
E0916 13:14:44.694734       1 accelerator.go:154] [DUMMY] doesn't contain GPU
E0916 13:14:44.694755       1 exporter.go:154] failed to init GPU accelerators: no devices found
I0916 13:14:44.697485       1 exporter.go:84] Number of CPUs: 16
I0916 13:14:45.420863       1 watcher.go:83] Using in cluster k8s config
I0916 13:14:45.522134       1 watcher.go:229] k8s APIserver watcher was started
I0916 13:14:45.524130       1 process_energy.go:129] Using the Ratio Power Model to estimate PROCESS_TOTAL Power
I0916 13:14:45.524176       1 process_energy.go:130] Feature names: [bpf_cpu_time_ms]
I0916 13:14:45.526583       1 process_energy.go:129] Using the Ratio Power Model to estimate PROCESS_COMPONENTS Power
I0916 13:14:45.526615       1 process_energy.go:130] Feature names: [bpf_cpu_time_ms bpf_cpu_time_ms bpf_cpu_time_ms   gpu_compute_util]
I0916 13:14:45.528749       1 node_component_energy.go:62] Skipping creation of Node Component Power Model since the system collection is supported
I0916 13:14:45.528912       1 prometheus_collector.go:95] Registered Container Prometheus metrics
I0916 13:14:45.529000       1 prometheus_collector.go:100] Registered VM Prometheus metrics
I0916 13:14:45.529072       1 prometheus_collector.go:104] Registered Node Prometheus metrics
I0916 13:14:45.534696       1 exporter.go:194] starting to listen on 0.0.0.0:9102
I0916 13:14:45.534832       1 exporter.go:208] Started Kepler in 845.25581ms
```
</details>

maryamtahhan, 2024-09-18:
This should be fixed now on the latest-dcgm image: https://github.com/sustainable-computing-io/kepler/actions/runs/10918073296/job/30302717983#step:7:1094

marvin-steinke, 2024-09-18:
> This should be fixed now on the latest-dcgm image: https://github.com/sustainable-computing-io/kepler/actions/runs/10918073296/job/30302717983#step:7:1094

Hi! I just tried:
```console
helm repo update
helm install kepler kepler/kepler --namespace kepler --create-namespace --set image.tag="latest-dcgm"
```
but
<details>
<summary>latest-dcgm</summary>

```console
E0918 11:33:58.736966       1 dcgm.go:66] Error initializing dcgm: Error connecting to nv-hostengine: Bad parameter passed to function
I0918 11:33:58.737134       1 exporter.go:103] Kepler running on version: v0.7.11-230-g600235fe-dirty
I0918 11:33:58.737184       1 config.go:293] using gCgroup ID in the BPF program: true
I0918 11:33:58.737203       1 config.go:295] kernel version: 6.8
I0918 11:33:58.737298       1 config.go:322] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0918 11:33:58.737343       1 power.go:59] use sysfs to obtain power
I0918 11:33:58.737355       1 redfish.go:169] failed to get redfish credential file path
I0918 11:33:58.742002       1 power.go:79] using acpi to obtain power
E0918 11:33:58.742030       1 accelerator.go:154] [DUMMY] doesn't contain GPU
E0918 11:33:58.742058       1 exporter.go:154] failed to init GPU accelerators: no devices found
I0918 11:33:58.744424       1 exporter.go:84] Number of CPUs: 16
I0918 11:33:59.341523       1 watcher.go:83] Using in cluster k8s config
I0918 11:33:59.443365       1 watcher.go:229] k8s APIserver watcher was started
I0918 11:33:59.446314       1 process_energy.go:129] Using the Ratio Power Model to estimate PROCESS_TOTAL Power
I0918 11:33:59.446368       1 process_energy.go:130] Feature names: [bpf_cpu_time_ms]
I0918 11:33:59.449626       1 process_energy.go:129] Using the Ratio Power Model to estimate PROCESS_COMPONENTS Power
I0918 11:33:59.449652       1 process_energy.go:130] Feature names: [bpf_cpu_time_ms bpf_cpu_time_ms bpf_cpu_time_ms   gpu_compute_util]
I0918 11:33:59.452812       1 node_component_energy.go:62] Skipping creation of Node Component Power Model since the system collection is supported
I0918 11:33:59.452942       1 prometheus_collector.go:95] Registered Container Prometheus metrics
I0918 11:33:59.453065       1 prometheus_collector.go:100] Registered VM Prometheus metrics
I0918 11:33:59.453122       1 prometheus_collector.go:104] Registered Node Prometheus metrics
I0918 11:33:59.459488       1 exporter.go:194] starting to listen on 0.0.0.0:9102
I0918 11:33:59.459662       1 exporter.go:208] Started Kepler in 722.553556ms
```
</details>

DCGM is again installed on the cluster through nvidia's `gpu-operator" helm chart and also directly on the host as a systemd service as [instructed by nvidia](https://developer.nvidia.com/dcgm).

Does NVML or DCGM work on your machine? I so, where should the libraries be located? Maybe my server is just configured weird.

> When I manually add `/usr/lib` (where nvidia-nvml is located) as a volume from the host to kepler's daemon set, kepler (or rather the go bindings library) finds the library and uses it to obtain gpu power.
> updated logs
> 
> I don't think this is the intended way. Did I install the driver or the container toolkit wrong?
> 
> ```
> $ sudo find / -name libnvidia-ml.so
> /usr/lib/x86_64-linux-gnu/libnvidia-ml.so
> /usr/lib/x86_64-linux-gnu/stubs/libnvidia-ml.so
> /var/lib/rancher/k3s/agent/containerd/io.containerd.snapshotter.v1.overlayfs/snapshots/1000/fs/usr/local/cuda-11.8/targets/x86_64-linux/lib/stubs/libnvidia-ml.so
> /var/lib/rancher/k3s/agent/containerd/io.containerd.snapshotter.v1.overlayfs/snapshots/788/fs/usr/local/cuda-11.8/targets/x86_64-linux/lib/stubs/libnvidia-ml.so
> /var/lib/rancher/k3s/agent/containerd/io.containerd.snapshotter.v1.overlayfs/snapshots/682/fs/usr/local/cuda-11.8/targets/x86_64-linux/lib/stubs/libnvidia-ml.so
> ```

maryamtahhan, 2024-09-18:
@marvin-steinke apologies, please bear with me - the configuration code was going through a lot of refactor in the last few weeks

There is now a different issue initializing dcgm (based on the log you provided) which I've just pushed a PR for: https://github.com/sustainable-computing-io/kepler/pull/1788

I'm waiting for a system to test on (will have one next week). But if you want to try to build a kepler image with that PR and see if it works in the meantime - please fire ahead.

maryamtahhan, 2024-09-27:
Testing https://github.com/sustainable-computing-io/kepler/pull/1788 ATM - will have an update shortly

closed 2024-09-30 by maryamtahhan