sustainable-computing-io/kepler #1784: kepler exporter pod stuck at ContainerCreating

smaitra53, 2024-09-13:
### What happened?

I have an rke cluster where I am trying to deploy kepler exporter using the `Deploy using Manifests` option. I have followed the [official documentation](https://sustainable-computing.io/installation/kepler/) for installation which is as follows:
```
git clone --depth 1 git@github.com:sustainable-computing-io/kepler.git
cd ./kepler
make build-manifest OPTS="PROMETHEUS_DEPLOY"
kubectl apply -f _output/generated-manifest/deployment.yaml
```

running `kubectl describe pods -n kepler` 
```
MountVolume.SetUp failed for volume "lib-modules" : hostPath type check failed: /lib/modules is not a directory
```
but the directory does exist on the host

### What did you expect to happen?

I expected the kepler exporter to run normally

### How can we reproduce it (as minimally and precisely as possible)?

- setup rke cluster with one node
- deploy prometheus and grafana as instructed in the official documentation
- deploy kepler as instructed in the official documentation

### Anything else we need to know?

Running the `kubectl describe pods -n kepler` command gives:

Name:           kepler-exporter-ggnx6
Namespace:      kepler
Priority:       0
Node:           192.168.1.119/192.168.1.119
Start Time:     Fri, 13 Sep 2024 13:07:35 -0400
Labels:         app.kubernetes.io/component=exporter
                app.kubernetes.io/name=kepler-exporter
                controller-revision-hash=5c97758dcd
                pod-template-generation=1
                sustainable-computing.io/app=kepler
Annotations:    <none>
Status:         Pending
IP:
IPs:            <none>
Controlled By:  DaemonSet/kepler-exporter
Containers:
  kepler-exporter:
    Container ID:
    Image:         quay.io/sustainable_computing_io/kepler:latest
    Image ID:
    Port:          9102/TCP
    Host Port:     0/TCP
    Command:
      /bin/sh
      -c
    Args:
      /usr/bin/kepler -v=1 -redfish-cred-file-path=/etc/redfish/redfish.csv
    State:          Waiting
      Reason:       ContainerCreating
    Ready:          False
    Restart Count:  0
    Requests:
      cpu:     100m
      memory:  400Mi
    Liveness:  http-get http://:9102/healthz delay=10s timeout=10s period=60s #success=1 #failure=5
    Environment:
      NODE_IP:     (v1:status.hostIP)
      NODE_NAME:   (v1:spec.nodeName)
    Mounts:
      /etc/kepler/kepler.config from cfm (ro)
      /etc/redfish from redfish (ro)
      /lib/modules from lib-modules (ro)
      /proc from proc (rw)
      /sys from tracing (ro)
      /var/run from var-run (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from kube-api-access-44pfj (ro)
Conditions:
  Type              Status
  Initialized       True
  Ready             False
  ContainersReady   False
  PodScheduled      True
Volumes:
  lib-modules:
    Type:          HostPath (bare host directory volume)
    Path:          /lib/modules
    HostPathType:  Directory
  tracing:
    Type:          HostPath (bare host directory volume)
    Path:          /sys
    HostPathType:  Directory
  proc:
    Type:          HostPath (bare host directory volume)
    Path:          /proc
    HostPathType:  Directory
  var-run:
    Type:          HostPath (bare host directory volume)
    Path:          /var/run
    HostPathType:  Directory
  cfm:
    Type:      ConfigMap (a volume populated by a ConfigMap)
    Name:      kepler-cfm
    Optional:  false
  redfish:
    Type:        Secret (a volume populated by a Secret)
    SecretName:  redfish-4kh9d7bc7m
    Optional:    false
  kube-api-access-44pfj:
    Type:                    Projected (a volume that contains injected data from multiple sources)
    TokenExpirationSeconds:  3607
    ConfigMapName:           kube-root-ca.crt
    ConfigMapOptional:       <nil>
    DownwardAPI:             true
QoS Class:                   Burstable
Node-Selectors:              <none>
Tolerations:                 node-role.kubernetes.io/control-plane:NoSchedule op=Exists
                             node-role.kubernetes.io/master:NoSchedule op=Exists
                             node.kubernetes.io/disk-pressure:NoSchedule op=Exists
                             node.kubernetes.io/memory-pressure:NoSchedule op=Exists
                             node.kubernetes.io/not-ready:NoExecute op=Exists
                             node.kubernetes.io/pid-pressure:NoSchedule op=Exists
                             node.kubernetes.io/unreachable:NoExecute op=Exists
                             node.kubernetes.io/unschedulable:NoSchedule op=Exists
Events:
  Type     Reason       Age                  From               Message
  ----     ------       ----                 ----               -------
  Normal   Scheduled    23m                  default-scheduler  Successfully assigned kepler/kepler-exporter-ggnx6 to 192.168.1.119
  Warning  FailedMount  12m (x2 over 19m)    kubelet            Unable to attach or mount volumes: unmounted volumes=[lib-modules], unattached volumes=[lib-modules tracing proc var-run cfm redfish kube-api-access-44pfj]: timed out waiting for the condition
  Warning  FailedMount  10m                  kubelet            Unable to attach or mount volumes: unmounted volumes=[lib-modules], unattached volumes=[var-run cfm redfish kube-api-access-44pfj lib-modules tracing proc]: timed out waiting for the condition
  Warning  FailedMount  7m46s (x2 over 16m)  kubelet            Unable to attach or mount volumes: unmounted volumes=[lib-modules], unattached volumes=[proc var-run cfm redfish kube-api-access-44pfj lib-modules tracing]: timed out waiting for the condition
  Warning  FailedMount  5m31s (x2 over 14m)  kubelet            Unable to attach or mount volumes: unmounted volumes=[lib-modules], unattached volumes=[cfm redfish kube-api-access-44pfj lib-modules tracing proc var-run]: timed out waiting for the condition
  Warning  FailedMount  3m16s                kubelet            Unable to attach or mount volumes: unmounted volumes=[lib-modules], unattached volumes=[tracing proc var-run cfm redfish kube-api-access-44pfj lib-modules]: timed out waiting for the condition
  Warning  FailedMount  60s (x19 over 23m)   kubelet            MountVolume.SetUp failed for volume "lib-modules" : hostPath type check failed: /lib/modules is not a directory
  Warning  FailedMount  59s (x2 over 21m)    kubelet            Unable to attach or mount volumes: unmounted volumes=[lib-modules], unattached volumes=[kube-api-access-44pfj lib-modules tracing proc var-run cfm redfish]: timed out waiting for the condition

### Kepler image tag

All image versions
```
quay.io/prometheus-operator/prometheus-config-reloaderv0.76.2
quay.io/prometheus-operator/prometheus-operatorv0.76.2
quay.io/brancz/kube-rbac-proxyv0.18.1
quay.io/prometheus/prometheusv2.54.1
quay.io/prometheus/node-exporterv1.8.2
quay.io/prometheus/blackbox-exporterv0.25.0
quay.io/prometheus/alertmanagerv0.27.0
```

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>
Client Version: version.Info{Major:"1", Minor:"23", GitVersion:"v1.23.5", GitCommit:"c285e781331a3785a7f436042c65c5641ce8a9e9", GitTreeState:"clean", BuildDate:"2022-03-16T15:58:47Z", GoVersion:"go1.17.8", Compiler:"gc", Platform:"linux/
amd64"}
Server Version: version.Info{Major:"1", Minor:"26", GitVersion:"v1.26.8", GitCommit:"395f0a2fdc940aeb9ab88849e8fa4321decbf6e1", GitTreeState:"clean", BuildDate:"2023-08-24T00:43:07Z", GoVersion:"go1.20.7", Compiler:"gc", Platform:"linux/
amd64"}

### Cloud provider or bare metal

<details>
bare metal
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.6 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.6 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
$ uname -a
Linux xxxx-PowerEdge-xxxx 5.15.0-119-generic #129~20.04.1-Ubuntu SMP Wed Aug 7 13:07:13 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE}
NAME         DATA   AGE
kepler-cfm   16     21m

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE}
Error from server (NotFound): deployments.apps "kepler-exporter" not found
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

dave-tucker, 2024-09-14:
This is due to https://github.com/sustainable-computing-io/kepler/blob/ea93777ad09a1e7f12c8dd1d9a97a61ce162e6d6/manifests/k8s/config/exporter/exporter.yaml#L89-L92

AFAICT this volume mount can be removed since it's not used anywhere.

smaitra53, 2024-09-16:
Thank you! Removing seems to have fixed the issue.

closed 2024-09-16 by smaitra53