sustainable-computing-io/kepler #1799: Kepler triggers eviction manager to reclaim ephemeral-storage

JonJon-Esto, 2024-09-26:
### What happened?

At some point during the 24 hours running kepler, the imagefs.available went below the threshold of 15% in a matter of 8-9 minutes.

38580  ---------------------------------------------------------------------
38581  1/1 | team
38582  ---------------------------------------------------------------------
38583  -- Current State ----------------------------------------------------
38584  ---------------------------------------------------------------------
38585  1. memory.available:      4902.88Mi
38586  2. nodefs.available:      8.00% [16.80/193.64Gi]
38587  3. nodefs.inodesFree:     98.00% [25428939/25804800]
**38588  4. imagefs.available:     8.00% [16.80/193.64Gi]**
38589  5. imagefs.inodesFree:    98.00% [25428939/25804800]
38590  6. pid.available:         62168
38591  ---------------------------------------------------------------------
38592  -- Eviction Thresholds ----------------------------------------------
38593  ---------------------------------------------------------------------
38594  1. memory.available:      100Mi
38595  2. nodefs.available:      10%
38596  3. nodefs.inodesFree:     5%
**38597  4. imagefs.available:     15%**
38598  5. imagefs.inodesFree:    5%
38599  6. pid.available:         null
38600  ---------------------------------------------------------------------

The ephemeral-storage seems to have degraded really fast without good enough log information about it. We can see from 05:03:07 today:

37669 ==================================================================================
37670 05:03:07 | INFO | Eviction signals for all nodes
37671 ==================================================================================
....
....
37718  /var/lib utilization:
37719 Filesystem      Size  Used Avail Use% Mounted on
37720 /dev/vda1       194G   30G  164G  16% /
37721  ---------------------------------------------------------------------

to 05:14:02:

38572 ==================================================================================
38573 05:14:02 | INFO | Eviction signals for all nodes
38574 ==================================================================================
....
....
38621  /var/lib utilization:
38622 Filesystem      Size  Used Avail Use% Mounted on
38623 /dev/vda1       194G  178G   16G  92% /
38624  ---------------------------------------------------------------------

Basically, disk utilization went up from 16% to 92% in just ~9mins.

This happened after installing kepler on top of my kubernetes setup.

**Setup description:**
09:01:09-estooald@team:-:~:kubectl get pods --all-namespaces
NAMESPACE     NAME                                   READY   STATUS    RESTARTS         AGE
kepler        kepler-exporter-lkzdj                  1/1     Running   0                30h
kube-system   coredns-7c65d6cfc9-8bwcp               1/1     Running   0                6d19h
kube-system   coredns-7c65d6cfc9-clbhr               1/1     Running   0                6d19h
kube-system   etcd-team                              1/1     Running   0                6d19h
kube-system   kube-apiserver-team                    1/1     Running   0                6d19h
kube-system   kube-controller-manager-team           1/1     Running   13 (3h53m ago)   6d19h
kube-system   kube-proxy-vkthn                       1/1     Running   0                6d19h
kube-system   kube-scheduler-team                    1/1     Running   13 (3h53m ago)   6d19h
monitoring    alertmanager-main-0                    2/2     Running   0                40m
monitoring    alertmanager-main-1                    2/2     Running   0                40m
monitoring    alertmanager-main-2                    2/2     Running   0                40m
monitoring    blackbox-exporter-86745676c9-pdx9j     3/3     Running   0                40m
monitoring    grafana-785fb96d65-mcq5h               1/1     Running   0                40m
[... 265 lines cut ...]
    DownwardAPI:             true
QoS Class:                   Burstable
Node-Selectors:              <none>
Tolerations:                 node-role.kubernetes.io/control-plane:NoSchedule op=Exists
                             node-role.kubernetes.io/master:NoSchedule op=Exists
                             node.kubernetes.io/disk-pressure:NoSchedule op=Exists
                             node.kubernetes.io/memory-pressure:NoSchedule op=Exists
                             node.kubernetes.io/not-ready:NoExecute op=Exists
                             node.kubernetes.io/pid-pressure:NoSchedule op=Exists
                             node.kubernetes.io/unreachable:NoExecute op=Exists
                             node.kubernetes.io/unschedulable:NoSchedule op=Exists
Events:                      <none>
</details>

### Container runtime (CRI) and version (if applicable)

<details>
ii  cri-tools                            1.31.1-1.1                              amd64        Command-line utility for interacting with a container runtime
</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>
ii  kubernetes-cni                       1.5.1-1.1                               amd64        Binaries required to provision kubernetes container networking
</details>

JonJon-Esto, 2024-09-28:
I have done an isolation test by removing kepler app and retained prometheus monitoring and observed that the eviction issue continues to happen.

Sep 28 05:37:59 team kubelet[2865036]: I0928 05:37:59.659033 2865036 eviction_manager.go:369] "Eviction manager: attempting to reclaim" resourceName="ephemeral-storage"

The setup now looks like:

Every 2.0s: kubectl get pods --all-namespaces                                                 team: Sat Sep 28 08:57:39 2024

NAMESPACE     NAME                                   READY   STATUS                   RESTARTS      AGE
kube-system   coredns-7c65d6cfc9-jw2ck               1/1     Running                  0             22h
kube-system   coredns-7c65d6cfc9-krzq2               1/1     Running                  0             22h
kube-system   etcd-team                              1/1     Running                  8             22h
kube-system   kube-apiserver-team                    1/1     Running                  7 (8h ago)    22h
kube-system   kube-controller-manager-team           1/1     Running                  24 (8h ago)   22h
kube-system   kube-proxy-cwtfx                       1/1     Running                  0             22h
kube-system   kube-scheduler-team                    1/1     Running                  24 (8h ago)   22h
monitoring    alertmanager-main-0                    2/2     Running                  0             18h
monitoring    alertmanager-main-1                    2/2     Running                  0             18h
monitoring    alertmanager-main-2                    2/2     Running                  0             18h
monitoring    blackbox-exporter-86745676c9-25xrm     3/3     Running                  0             18h
monitoring    grafana-785fb96d65-8qh8s               0/1     ContainerStatusUnknown   1             18h
monitoring    grafana-785fb96d65-dcbcg               1/1     Running                  0             3h17m
monitoring    kube-state-metrics-6f46974967-c7fkk    3/3     Running                  0             18h
monitoring    node-exporter-z8f8n                    2/2     Running                  0             18h
monitoring    prometheus-adapter-784f566c54-9688f    1/1     Running                  0             18h
monitoring    prometheus-adapter-784f566c54-x4h52    1/1     Running                  0             18h
monitoring    prometheus-k8s-0                       2/2     Running                  0             3h18m
monitoring    prometheus-k8s-1                       2/2     Running                  0             3h17m
monitoring    prometheus-operator-57b579d5b9-gwmk9   2/2     Running                  0             18h

So, it seems that prometheus is causing the eviction issue.

Anyone who can help in digging further as to why this is happening? Thanks!

JonJon-Esto, 2024-09-30:
Hi,

Please close this ticket. I have managed to isolate the Kepler and prometheus system by removing them and leaving the kubernetes base system on its own. Still, eviction manager was triggered. Something else is taking up huge amount of disk space somewhere in the system.

Thank you!

sthaha, 2024-10-14:
closing the issue based on https://github.com/sustainable-computing-io/kepler/issues/1799#issuecomment-2382156762