sustainable-computing-io/kepler #1806: Kepler process / vm bpf cpu time jitters quite a lot compare to node-exporter cpu time

sthaha, 2024-10-14:
Kepler Version: 
* 
kepler_exporter_build_info{arch="amd64", branch="main", instance="vm:9100", job="vm", os="linux", revision="8672dc3cf70a20bdadb5f8680699e812ef0998f5", version="v0.7.11-254-g8672dc3c-dirty"} | 1
* kepler_exporter_build_info{arch="amd64", branch="main", instance="kepler-metal:8888", job="metal", os="linux", revision="8672dc3cf70a20bdadb5f8680699e812ef0998f5", version="v0.7.11-254-g8672dc3c-dirty"}

Steps to reproduce 
* Run any dev or validation/metal or validation/vm [compose for kepler ](https://github.com/sustainable-computing-io/kepler/tree/main/manifests/compose)
* Run stress-ng that stresses the machine for some time
* Open Prometheus and run `sum(rate(kepler_process_bpf_cpu_time_ms_total{job="metal"}[20s]))`

See the jitter as below 
[image]
[image]

### Node Exporter CPU time vs Kepler 
* node-exporter in green is almost a straight line 
* sum(rate(kepler_vm_package_joules_total{job="metal"}[20s]))  in red 
* sum(rate(kepler_process_bpf_cpu_time_ms_total{job="vm" pid!="1"}[20s])) in yellow

[image]

Zoomed: 

[image]

vimalk78, 2024-10-17:
can this be closed?

sthaha, 2024-10-18:
Yes, this jitter happens if the scrape interval is less than kepler's update interval of 3s

closed 2024-10-18 by sthaha