sustainable-computing-io/kepler #1811: Disable Idle/Dynamic Separation in Kepler

KaiyiLiu1234, 2024-10-16:
Add flag to allows users to disable Idle/Dynamic Separation in Kepler, so Kepler only exports direct power measurements. Idle/Dynamic Separation is faulty and also complicates the validation process.

aharivel, 2024-10-23:
This is an interesting statement. Could you elaborate why this is faulty ? Overall I agree with the fact that it is easier to only have one value.

KaiyiLiu1234, 2024-10-23:
From a previous test, idle power never goes down. It seems to always select the maximum "total power" and sticks to it. So what happens is if you stress the cpu to 100%, idle will adopt this power and sticks to it. When the stressing stops, the total power from rapl goes down but idle assigned by kepler does not change. When dynamic is calculated with max(total power - idle power, 0), dynamic is stuck at 0 after the first stress and idle is stuck at the highest power level.

KaiyiLiu1234, 2024-10-23:
Flag has been added so the issue can be closed.

closed 2024-10-23 by KaiyiLiu1234

aharivel, 2024-10-24:
Could you give the link to the PR that add the flag please ? Thanks