sustainable-computing-io/kepler #1897: Kepler node platform metric not available on BM

vprashar2929, 2024-12-20:
### What happened?

In case of `uncore` rapl domain available on a BM machine Kepler doesn't produce `kepler_node_platform` metric

Ref:
[image]

[image]

[image]

### What did you expect to happen?

Since `kepler_node_platform` metric represents the total energy consumption of the host https://github.com/sustainable-computing-io/kepler-doc/blob/main/docs/design/metrics.md we should see the metric available irrespective of uncore rapl domain available or not represents

### How can we reproduce it (as minimally and precisely as possible)?

Use a BM machine where uncore rapl domain is available. To check what all domains are available use below command:
```
for file in $(sudo find -L /sys/class/powercap/intel-rapl -name name  2>/dev/null); do cat $file;  done  | sort -n| uniq
```

### Kepler image tag

<details>
latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>
Bare metal
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE}
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE}
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

vprashar2929, 2025-01-07:
cc: @rootfs @KaiyiLiu1234 https://github.com/sustainable-computing-io/kepler-metal-ci/issues/335#issuecomment-2523846396 I also see the same weird behaviour in our MSE and MAPE numbers as well in case when `uncore` rapl domain is available

sthaha, 2025-08-11:
Closing this as not longer relevant in the new version.

closed 2025-08-11 by sthaha