sustainable-computing-io/kepler #1905: [CI] Using AI to generate unit test code and improve memory leak?

SamYuan1990, 2025-01-07:
Learn from kepler-metal-ci's report analysis, I want to proposal an new approach for us to improve our test code coverage and fixing memory leaks.
For now, we have our test coverage in report by golang command as `... --cover` and memory leak detection as `go build -xxx`, which both are plain text result.
so
if we make a python script, reading those report, and 
## for unit test code generation.
```
filter by unit test result
- if this package is not on ban list(for example, version package do not need to test?)
- if some package's test coverage lower than a threshold.
- we send files inside this package to openAI to generate test result.
- we recevie the test code generate by openAI
- we rerun the test coverage with test files, if it improve test coverage for the package.
- create a new PR.
```

## for memory leak detection
```
for each PR, filter by memory leak result
- if the issue is on ban list(for example, string or big slice/array moved to heap)
- if a new memory leak happen
- send the specific leak to openAI to see if this can be fixed or not?
- if new memory leak been added or the memory leak can be fixed as AI's suggestion.
- leave comments in PR for double check/manual review.
```

sunya-ch, 2025-01-07:
We do have a way to enable pprof in Kepler: https://github.com/sustainable-computing-io/kepler/blob/main/doc/dev/README.md#profiling-kepler. 
I think we can utilize it to spot the memory leak.

SamYuan1990, 2025-01-07:
> We do have a way to enable pprof in Kepler: https://github.com/sustainable-computing-io/kepler/blob/main/doc/dev/README.md#profiling-kepler. I think we can utilize it to spot the memory leak.

IMO, golang has two approaches for memory enhancement, one is pprof, and another one is escapes detect like https://github.com/sustainable-computing-io/kepler/actions/runs/12637476502/job/35211752748

closed 2026-05-07 by github-actions[bot]