sustainable-computing-io/kepler #1926: Kepler underreporting power usage for gpu_joules_total

vprashar2929, 2025-02-17:
### What happened?

Kepler when deployed using `release-0.7.10` (With/Without dcgm) and `release-0.7.11` (With/Without dcgm) it under reports the Power usage when compared against the DCGM exporter that exposes [DCGM_FI_DEV_POWER_USAGE](https://docs.nvidia.com/datacenter/dcgm/latest/dcgm-api/dcgm-api-field-ids.html#c.DCGM_FI_DEV_POWER_USAGE) which tells you power usage in Watts. 

When using `release-0.7.10` Kepler uses NVML

[image]

[image]

When using `release-0.7.10-dcgm` Kepler uses DCGM

[image]

When using `release-0.7.11-dcgm` Kepler uses DCGM

[image]

### What did you expect to happen?

It should'nt produce a lower value than what DCGM exporter produces

### How can we reproduce it (as minimally and precisely as possible)?

1. Deploy Kepler on cluster with the mentioned versions that have access to the GPU
2. Stress the GPU nodes so that we can get some value

### Kepler image tag

<details>
release-0.7.10, release-0.7.10-dcgm, release-0.7.11, release-0.7.11-dcgm
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>
VM
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE}
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE}
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

vprashar2929, 2025-02-17:
@rootfs @sunya-ch @marceloamaral Could anyone provide insights on how Kepler's GPU metrics were validated against DCGM/NVML? Additionally, were there any alternative validation methods used when these features were initially introduced in Kepler?

sthaha, 2025-08-11:
Not supported in Kepler rewrite.

closed 2025-08-11 by sthaha