sustainable-computing-io/kepler #1932: Per Socket/Package Resource Allocation/CPU Time is not available

KaiyiLiu1234, 2025-02-25:
### What happened?

Kepler provides idle power per socket as shown here: https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/stats/node_stats.go#L72. However, the cpu time is for the entire node as shown here: https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/stats/node_stats.go#L69. I believe Kepler requires per socket cpu time to calculate the per socket idle power, especially now that we intend to modify Kepler to use a Linear Model to predict idle power. 

### What did you expect to happen?

Kepler should instead provide resource allocation at a per socket/package level to be used for idle power calculation.

### How can we reproduce it (as minimally and precisely as possible)?

Run Kepler Go Binary with --expose-estimated-idle-power=true

### Kepler image tag

<details>
latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE}
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE}
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

sunya-ch, 2025-03-05:
Related issue in the past: https://github.com/sustainable-computing-io/kepler/issues/570

sthaha, 2025-08-11:
Closing this as not longer relevant in the new version.

closed 2025-08-11 by sthaha