sustainable-computing-io/kepler #1939: kernel_processes/kernel/PKG and kernel_processes/kernel/DRAM have the highest power consumptions

AntonioDiTuri, 2025-03-04:
### What happened?

I installed kepler using the helmchart version 0.5.13 and I am visualizing the data of the Grafana Dashboard in my dev cluster. I am running kepler on a EKS cluster in AWS (more information about the technical setting below in the how to reproduce step

### What did you expect to happen?

I expected to analyze data coming from different namespaces from my dev cluster. 
Unfortunately I noticed that most of the energy consumption was coming from the "kernel" namespace.

First question: what is the kernel namespace in kepler internals? 
My assumption is that this is the energy related to the kepler interaction with the kernel.

Is it expected to be way higher than the actual application namespaces?

I have round about 250 pods, not super active but it feels wrong that the kernel namespace has so much higher values in terms of power and related energy

I am attaching a screen from your grafana dashboard.

[image]

### How can we reproduce it (as minimally and precisely as possible)?

Set up an EKS cluster with k8s version 1.30.
Use a managed node group.
AMI: amazon-eks-node-1.30-v20250212
container runtime: containerd://1.7.25
kernel version: 5.10.233-224.894.amzn2.x86_64

Install kepler using helm chart v.0.5.13

Install some random test application to confront the values. (maybe this passage can be skipped).
Check the grafana dashboard

### Anything else we need to know?

Please let me know if some information is missing

### Kepler image tag

<details>
From the chart appReleaseVersion is: [release-0.7.12](https://github.com/sustainable-computing-io/kepler-helm-chart/blob/8aae20112f0b7503405d604509a1bd2e2566c600/chart/kepler/Chart.yaml#L26) .

However when I describe the pod I see that the version taken is: sustainable_computing_io/kepler:latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
Client Version: v1.32.0
```

</details>

### Cloud provider or bare metal

<details>
AWS
Set up an EKS cluster with k8s version 1.30.
Use a managed node group.
AMI: amazon-eks-node-1.30-v20250212
container runtime: containerd://1.7.25
kernel version: 5.10.233-224.894.amzn2.x86_64
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
NAME="Amazon Linux"
VERSION="2"
ID="amzn"
ID_LIKE="centos rhel fedora"
VERSION_ID="2"
PRETTY_NAME="Amazon Linux 2"
ANSI_COLOR="0;33"
CPE_NAME="cpe:2.3:o:amazon:amazon_linux:2"
HOME_URL="https://amazonlinux.com/"
SUPPORT_END="2026-06-30"
$ uname -a
Linux ip-XXX-XXX-XXX-XXX.availability_zone.compute.internal 5.10.233-224.894.amzn2.x86_64 #1 SMP Mon Jan 27 16:52:48 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE}
The helm chart has no configmap
# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE}
# attaching the daemonset descripton: 

Name:           kepler
Selector:       app.kubernetes.io/component=exporter,app.kubernetes.io/name=kepler
Node-Selector:  kubernetes.io/os=linux
Labels:         app.kubernetes.io/component=exporter
                app.kubernetes.io/managed-by=Helm
                app.kubernetes.io/name=kepler
                app.kubernetes.io/version=release-0.7.12
                helm.sh/chart=kepler-0.5.13
Annotations:    deprecated.daemonset.template.generation: 1
                meta.helm.sh/release-name: kepler
                meta.helm.sh/release-namespace: kepler
Desired Number of Nodes Scheduled: 10
Current Number of Nodes Scheduled: 10
Number of Nodes Scheduled with Up-to-date Pods: 10
Number of Nodes Scheduled with Available Pods: 10
Number of Nodes Misscheduled: 0
Pods Status:  10 Running / 0 Waiting / 0 Succeeded / 0 Failed
Pod Template:
  Labels:           app.kubernetes.io/component=exporter
                    app.kubernetes.io/name=kepler
  Service Account:  kepler-sa
  Containers:
   kepler-exporter:
    Image:      custom-artifactory/sustainable_computing_io/kepler:release-0.7.12
    Port:       9102/TCP
    Host Port:  9102/TCP
    Args:
      -v=$(KEPLER_LOG_LEVEL)
    Liveness:  http-get http://:9102/healthz delay=10s timeout=10s period=60s #success=1 #failure=5
    Environment:
      NODE_IP:                      (v1:status.hostIP)
      NODE_NAME:                    (v1:spec.nodeName)
      METRIC_PATH:                 /metrics
      BIND_ADDRESS:                0.0.0.0:9102
      CGROUP_METRICS:              *
      CPU_ARCH_OVERRIDE:
      ENABLE_EBPF_CGROUPID:        true
      ENABLE_GPU:                  true
      ENABLE_PROCESS_METRICS:      false
      ENABLE_QAT:                  false
      EXPOSE_CGROUP_METRICS:       false
      EXPOSE_HW_COUNTER_METRICS:   true
      EXPOSE_IRQ_COUNTER_METRICS:  true
      KEPLER_LOG_LEVEL:            1
    Mounts:
      /lib/modules from lib-modules (rw)
      /proc from proc (rw)
      /sys from tracing (rw)
      /usr/src from usr-src (rw)
  Volumes:
   lib-modules:
    Type:          HostPath (bare host directory volume)
    Path:          /lib/modules
    HostPathType:  DirectoryOrCreate
   tracing:
    Type:          HostPath (bare host directory volume)
    Path:          /sys
    HostPathType:  Directory
   proc:
    Type:          HostPath (bare host directory volume)
    Path:          /proc
    HostPathType:  Directory
   usr-src:
    Type:          HostPath (bare host directory volume)
    Path:          /usr/src
    HostPathType:  Directory
  Node-Selectors:  kubernetes.io/os=linux
  Tolerations:     node-role.kubernetes.io/control-plane:NoSchedule
Events:
  Type    Reason            Age   From                  Message
  ----    ------            ----  ----                  -------
  Normal  SuccessfulCreate  17m   daemonset-controller  Created pod: kepler-clstx
  Normal  SuccessfulCreate  17m   daemonset-controller  Created pod: kepler-n6bsm
  Normal  SuccessfulCreate  17m   daemonset-controller  Created pod: kepler-t2mj9
  Normal  SuccessfulCreate  17m   daemonset-controller  Created pod: kepler-h7gft
  Normal  SuccessfulCreate  17m   daemonset-controller  Created pod: kepler-689b5
  Normal  SuccessfulCreate  17m   daemonset-controller  Created pod: kepler-5g49p
  Normal  SuccessfulCreate  17m   daemonset-controller  Created pod: kepler-s7zp7
  Normal  SuccessfulCreate  17m   daemonset-controller  Created pod: kepler-42w7l
  Normal  SuccessfulCreate  17m   daemonset-controller  Created pod: kepler-p97m8
  Normal  SuccessfulCreate  17m   daemonset-controller  (combined from similar events): Created pod: kepler-z884j
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

sunya-ch, 2025-03-04:
@AntonioDiTuri Thank you for the issue. It seems to be a static namespace to be used when the process has CGROUP=1.

https://github.com/sustainable-computing-io/kepler/blob/6fdba6eee1af2cc24762cd81873a65b5336d2335/pkg/cgroup/resolve_container.go#L247

@rootfs @marceloamaral  Are you aware of the change affected to this part?

SamYuan1990, 2025-03-04:
@AntonioDiTuri , could you please provide more logs from kepler container?

KaiyiLiu1234, 2025-03-04:
Instead of using 0.7.12 tag, can you replace the helm chart tag to "latest" and see if you get different results?

AntonioDiTuri, 2025-03-04:
@sunya-ch the kind of application I am running at the moment are not very cpu intensive. So the static namespace would be the one that takes the "kernel" name?

@KaiyiLiu1234 I tried and I have similar results. 

@SamYuan1990 here you are: 

```
WARNING: failed to read int from file: open /sys/devices/system/cpu/cpu0/online: no such file or directory
I0304 11:13:58.232435       1 exporter.go:103] Kepler running on version: v0.7.12-dirty
I0304 11:13:58.232505       1 config.go:293] using gCgroup ID in the BPF program: true
I0304 11:13:58.232527       1 config.go:295] kernel version: 5.1
I0304 11:13:58.232601       1 power.go:78] Unable to obtain power, use estimate method
I0304 11:13:58.232617       1 redfish.go:169] failed to get redfish credential file path
I0304 11:13:58.232839       1 acpi.go:71] Could not find any ACPI power meter path. Is it a VM?
I0304 11:13:58.232848       1 power.go:79] using none to obtain power
E0304 11:13:58.232856       1 accelerator.go:154] [DUMMY] doesn't contain GPU
E0304 11:13:58.232869       1 exporter.go:154] failed to init GPU accelerators: no devices found
WARNING: failed to read int from file: open /sys/devices/system/cpu/cpu0/online: no such file or directory
I0304 11:13:58.233751       1 exporter.go:84] Number of CPUs: 8
W0304 11:13:58.367060       1 exporter.go:299] Failed to open perf event for CPU cycles: failed to open bpf perf event on cpu 0: no such file or directory
I0304 11:13:58.368565       1 watcher.go:83] Using in cluster k8s config
I0304 11:13:58.469122       1 watcher.go:229] k8s APIserver watcher was started
I0304 11:13:58.470428       1 process_energy.go:129] Using the Ratio Power Model to estimate PROCESS_TOTAL Power
I0304 11:13:58.470449       1 process_energy.go:130] Feature names: [bpf_cpu_time_ms]
I0304 11:13:58.471586       1 process_energy.go:129] Using the Ratio Power Model to estimate PROCESS_COMPONENTS Power
I0304 11:13:58.471598       1 process_energy.go:130] Feature names: [bpf_cpu_time_ms bpf_cpu_time_ms bpf_cpu_time_ms   gpu_compute_util]
I0304 11:13:58.474008       1 regressor.go:276] Created predictor linear for trainer: "SGDRegressorTrainer"
I0304 11:13:58.474023       1 model.go:125] Requesting for Machine Spec: &{genuineintel intel_xeon_platinum_8375c 8 1 30 3500 2}
I0304 11:13:58.474033       1 node_platform_energy.go:53] Using the Regressor/AbsPower Power Model to estimate Node Platform Power
I0304 11:13:58.476296       1 regressor.go:276] Created predictor linear for trainer: "SGDRegressorTrainer"
I0304 11:13:58.476306       1 regressor.go:276] Created predictor linear for trainer: "SGDRegressorTrainer"
I0304 11:13:58.476309       1 regressor.go:276] Created predictor linear for trainer: "SGDRegressorTrainer"
I0304 11:13:58.476312       1 regressor.go:276] Created predictor linear for trainer: "SGDRegressorTrainer"
I0304 11:13:58.476316       1 model.go:125] Requesting for Machine Spec: &{genuineintel intel_xeon_platinum_8375c 8 1 30 3500 2}
I0304 11:13:58.476324       1 node_component_energy.go:57] Using the Regressor/AbsPower Power Model to estimate Node Component Power
I0304 11:13:58.476370       1 prometheus_collector.go:95] Registered Container Prometheus metrics
I0304 11:13:58.476397       1 prometheus_collector.go:100] Registered VM Prometheus metrics
I0304 11:13:58.476405       1 prometheus_collector.go:104] Registered Node Prometheus metrics
I0304 11:13:58.477680       1 exporter.go:194] starting to listen on 0.0.0.0:9102
I0304 11:13:58.477742       1 exporter.go:208] Started Kepler in 245.420153ms
```

AntonioDiTuri, 2025-03-06:
Any updates here? Just wanted to mention that I did similar test in production with same environment and this time with appliaction having high usage of CPU but I can observe similar pattern. 

The kernel namespace being the one consuming more than everyone else. 

Can someone provide an explenation about this kernel namespace?

Thanks in advance

SamYuan1990, 2025-03-06:
> Any updates here? Just wanted to mention that I did similar test in production with same environment and this time with appliaction having high usage of CPU but I can observe similar pattern.
>
> The [...]

For my understanding, well, I am major focus on CI...
from code block, as EnabledEBPFCgroupID there, the parameter should enable for kernel version > 4.18 in our case, we have 5.1 and cgroupv2.

https://github.com/sustainable-computing-io/kepler/blob/e8c14040cabcbddd1118cb62c84a2fe7aae0b812/pkg/config/config.go#L310-L319

I hope I am right as this setting down to 
[pkg/collector/metric_collector.go](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/metric_collector.go#L184)

https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/utils.go#L28-L37

and finally to
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/cgroup/resolve_container.go#L247-L254

set namespace to Kernel
ref to comments
// SetEnabledEBPFCgroupID enables or disables eBPF code to collect cgroup ID
// based on kernel version and cgroup version.
// SetEnabledEBPFCgroupID enables the eBPF code to collect cgroup id if the system has kernel version > 4.18

as I am not eBPF expert, if you want to have a try, @AntonioDiTuri maybe you can try to manual set the config to false.

sunya-ch, 2025-03-07:
@AntonioDiTuri cgroup ID = 1 means that this process is not belong to any cgroup. It is summation of process which is running outside the container cgroups.
We can further investigate which process causes this consumption by disabling the cgroup and report in process level.

AntonioDiTuri, 2025-03-08:
Thanks for the quick answer and sorry for the delay!

I have disabled the EBPFCgroupID using the environment variable: `ENABLE_EBPF_CGROUPID`.

However after that I have only 3 non zero metrics in the kernel namespace, all the other are zero.

[image]

I also tried, out of curiosity on a local setup using minikube on a mac with documentation installation and I have the same behavior

sunya-ch, 2025-03-12:
@AntonioDiTuri 
I think it is now using `kepler_container` query. Could you please try replacing with `kepler_process` prefix?

SamYuan1990, 2025-03-12:
@AntonioDiTuri , @KaiyiLiu1234 , @sunya-ch I have a ✨ idea just come in my mind.
https://github.com/sustainable-computing-io/kepler-metal-ci
as kepler community has a regression test in AWS, so maybe @AntonioDiTuri can fork it in your own, and config a CI targets on your AWS instance. if the kepler runs well with our CI, then I suppose it's config issue with minikube.(our CI today running on kind if I am correct.) for further test logic, I suppose @KaiyiLiu1234 can help brief.
I don't think we support mac today.(we just support compile in code level to make developer can contribute code as code fix with mac, as mac user as me...)
btw, @sunya-ch has  a call for machine propose, it's another approach for reuse our CI maybe?

Artiach, 2025-05-05:
Hi all,

I was having this same issue when running kepler on a kind cluster running on top of WSL2. The only non-zero container's consumption were coming from kernel namespace. 

After reading in this issue about the problem being related with the Cgroups I noticed that my WSL was running on a hybrid mode between cgroups v1 and v2. So I updated my WSL to run only cgroupsv2 (https://github.com/spurin/wsl-cgroupsv2), created a new kind cluster and deployed kepler again and seems to have been solved. 

I dont know if this solution fits in this issue but hope it can help.

locomundo, 2025-06-15:
Hi!

I'm also testing out Kepler on EKS and find similar behaviour. I was wondering if there was a conclusion about this.

I've been testing with more resource intensive software like was suggested, and then the proportion gets less significant.
I am running the [opentelemetry-demo application](https://github.com/open-telemetry/opentelemetry-demo), which includes some load generators for testing OTel.

See screenshots. In the first one I can see my application pods can appear in the top 10. In the second one, without the resource-intensive application running, `kernel_processes` are way above anything else.

Is this expected behaviour? Does it only happen in EKS? Is it because EKS is running controllers and other software in the background which consume a lot of resources (at least proportionally)?

Thank you for any ideas on this!

[image]

[image]