sustainable-computing-io/kepler #2032: Kepler utilizes 1 vCPU core out of 4 on small EKS clusters

nikimanoledaki, 2025-04-28:
In some small clusters, Kepler pods consistently utilize around 1 full vCPU core. The clusters are 4-core nodes, so the compute utilization rate of Kepler consumes 25% of the total compute power in these small clusters, which represents a significant cost of 25% for these small clusters.

The nodes are on EKS, specifically on C5 instance types. Please let me know if there is any information I could provide to help with this investigation.

I discussed this with @sunya-ch while at KubeCon in London. Is there anything that can be tweaked?

sunya-ch, 2025-04-28:
@nikimanoledaki Thank you! Could you please share Kepler version and also the configuration to us? I think we also have quota to run C5 on EC2 spot instance to reproduce the issue, aren't we @sthaha?

nikimanoledaki, 2025-04-28:
Thank you so much @sunya-ch! Adding more info below.

Kepler version: 0.7.11 (just realised that we'll need to upgrade to 0.8.0)

<details><summary>Kepler DaemonSet manifest</summary>

```
apiVersion: apps/v1
kind: DaemonSet
metadata:
  labels:
    app.kubernetes.io/component: exporter
    app.kubernetes.io/name: kepler
    app.kubernetes.io/version: release-0.7.8 # image used is actually 0.7.11 (see below)
    helm.sh/chart: kepler-0.5.6
  name: kepler
  namespace: kepler
spec:
  selector:
    matchLabels:
      app.kubernetes.io/component: exporter
      app.kubernetes.io/name: kepler
  template:
    metadata:
      labels:
        app.kubernetes.io/component: exporter
        app.kubernetes.io/name: kepler
        name: kepler
    spec:
      containers:
      - args:
        - -v=$(KEPLER_LOG_LEVEL)
        env:
        - name: NODE_IP
          valueFrom:
            fieldRef:
              fieldPath: status.hostIP
        - name: NODE_NAME
          valueFrom:
            fieldRef:
              fieldPath: spec.nodeName
        - name: METRIC_PATH
          value: /metrics
        - name: BIND_ADDRESS
          value: 0.0.0.0:9102
        - name: CGROUP_METRICS
          value: '*'
        - name: CPU_ARCH_OVERRIDE
          value: ""
        - name: ENABLE_EBPF_CGROUPID
          value: "true"
        - name: ENABLE_GPU
          value: "true"
        - name: ENABLE_PROCESS_METRICS
          value: "false"
        - name: ENABLE_QAT
          value: "false"
        - name: EXPOSE_CGROUP_METRICS
          value: "false"
        - name: EXPOSE_HW_COUNTER_METRICS
          value: "true"
        - name: EXPOSE_IRQ_COUNTER_METRICS
          value: "true"
        - name: KEPLER_LOG_LEVEL
          value: "1"
        - name: EXPOSE_ESTIMATED_IDLE_POWER_METRICS
          value: "true"
        image: quay.io/sustainable_computing_io/kepler:release-0.7.11
        imagePullPolicy: Always
        livenessProbe:
          failureThreshold: 5
          httpGet:
            path: /healthz
            port: 9102
            scheme: HTTP
          initialDelaySeconds: 10
          periodSeconds: 60
          successThreshold: 1
          timeoutSeconds: 10
        name: kepler-exporter
        ports:
        - containerPort: 9102
          hostPort: 9102
          name: http-metrics
        resources:
          limits:
            cpu: "2"
            memory: 2.5Gi
          requests:
            cpu: "0.01"
            memory: 100Mi
        securityContext:
          privileged: true
        volumeMounts:
        - mountPath: /lib/modules
          name: lib-modules
        - mountPath: /sys
          name: tracing
        - mountPath: /proc
          name: proc
      hostNetwork: true
      nodeSelector:
        kubernetes.io/os: linux
      priorityClassName: high
      serviceAccountName: kepler
      tolerations:
      - operator: Exists
      volumes:
      - hostPath:
          path: /lib/modules
          type: DirectoryOrCreate
        name: lib-modules
      - hostPath:
          path: /sys
          type: Directory
        name: tracing
      - hostPath:
          path: /proc
          type: Directory
        name: proc
```

</details>

nikimanoledaki, 2025-06-25:
I will try one of the version under the [`reboot` tag](https://github.com/sustainable-computing-io/kepler/releases?q=reboot&expanded=true) and let you know 🤞