sustainable-computing-io/kepler #2079: [reboot]: Support Platform Metrics / ACPI

sthaha, 2025-05-20:


ExplorerRay, 2025-08-11:
If I want to take this issue, should I write a proposal like [this](https://github.com/sustainable-computing-io/kepler/pull/2245/files)?

sthaha, 2025-08-11:
@ExplorerRay thank you for helping out. Yes, lets start with the proposal.

sthaha, 2025-08-11:
@ExplorerRay 

> In functional metrics, it says Platform power metrics available on 95%+ of nodes with BMCs, but how to determine the implemented Kepler is supported by so many nodes?

we can skip that in the proposal. In Redfish case, I was relying on gofish as a way to support reading from redfish. 

> For performance metrics, is there an existing way to test Kepler's performance? With it, all developers will have the same standard for how to judge the overhead. 

Ideally CI should give us the performance difference (cc: @vprashar2929 ) when a PR is added to kepler but comparing `main` vs `pr` after running both versions. But again, in ACPI's case unless we have the hardware, it may not even be possible.

ExplorerRay, 2025-08-15:
When considering the implementation of this feature, I am wondering why the Redfish PR uses new `internal/platform` directory instead of `internal/device`. I think it's because that the Redfish client is different from reading files in sysfs. Therefore, it's better to use a split directory which may make `internal/device` look cleaner. But for ACPI, it also read files in sysfs while it collects power data for platform. I am not sure which directory is better for ACPI power meter code.
For my opinion, I want to follow the Redfish PR because I think the ACPI power meter code won't reuse RAPL ones although it is actually a device. However, I don't know whether it's good or not to follow a PR that hasn't been merged. Will that change a lot before it merged?

There is another problem I want to check. The Redfish PR doesn't use original power monitor for power attribution. I think this is because it is hard to determine whole node attribution because there might be many devices that are not in procfs. If I’m mistaken, please correct me. Thanks.

sthaha, 2025-08-17:
Since its all internal packages (i.e not exported outside), we have the freedom to refactor it any point in time. Like you said, `redfish` isn't a device per se, so I didn't keep it inside device package. The idea behind `platform` (in my mind) is to align with the terminology `kepler` had chosen earlier and to clearly indicate that the power are reported at a node level and not process level. I am happy to move the package into device if that makes sense ... e.g `internal/device/platform` is equally good to me 

But again .. we can change structure or the packages when we have a clearer picture of the organisation. 

> There is another problem I want to check. The Redfish PR doesn't use original power monitor for power attribution. I think this is because it is hard to determine whole node attribution because there might be many devices that are not in procfs. If I’m mistaken, please correct me. 

No, you are right. We can't apportion **platform** power to processes solely based on cpu usage percentage.

ExplorerRay, 2025-08-26:
I just read the [metric part of Redfish proposal](https://github.com/sustainable-computing-io/kepler/blob/main/docs/developer/proposal/EP_001-redfish-support.md#metrics) which mentions only watt metric is provided makes me curious about the design.

Regarding the RAPL sysfs, it natively provides only energy (not power) information. I noticed that [some functions in Kepler](https://github.com/sustainable-computing-io/kepler/blob/main/internal/monitor/node.go#L10) calculate power from these energy metrics.

So, why not use the power metric from Redfish and multiply it by the elapsed time to obtain `joules_total`? Since RAPL seems to rely on a similar strategy, if we are concerned about the precision of this method, then perhaps the RAPL-derived power metric should also be reconsidered.

Since ACPI also natively provides only power metrics, I’d like to confirm this understanding.

sthaha, 2025-08-26:
@ExplorerRay It is fine to construct (average) power from  energy but not the other way because power reading is  instantaneous and not the average power (since the last read). 
Say ... 

| T(ms)   | Power | Energy |
|---------|-------|--------|
|* 0  |  20   | ?? |
| 99 |  40.  | |
|*100 |  40   | ?? |
|199 |  80   | ?? |
|*200 |  30   |   |
|280 |  10   |   |
|*300 |  30   |   |
| 390 |  150  |   |
|*400 |  30   |   |

And Power is measured at every 100s ... It would be wrong to use the instantaneous power to compute the energy consumed over the interval. Where as in the case of RAPL, because we are counting Energy (J), it is fine to compute the average power consumed as Δj/Δt

ExplorerRay, 2025-08-26:
Thanks for your detailed response for clarifying~

ACPI has exposed `power[1-*]_average` over `power[1-*]_average_interval`, so I think it is fine to use `power_average` multiplies `power_average_interval` for getting `joules_total` for platform-level energy metrics.

Ref: [https://www.kernel.org/doc/Documentation/hwmon/acpi_power_meter](https://www.kernel.org/doc/Documentation/hwmon/acpi_power_meter)