sustainable-computing-io/kepler #2126: Can't get GPU metrics

mohamedosama113, 2025-06-03:
I deployed Kepler Helm on GKE ubuntu nodes and  I these are logs of kepler pod 
```
WARNING: failed to read int from file: open /sys/devices/system/cpu/cpu0/online: no such file or directory
I0603 10:52:04.432230       1 dcgm.go:70] Initializing dcgm Successful
E0603 10:52:04.432325       1 device.go:103] Device with type NVML doesn't exist
I0603 10:52:04.432337       1 dcgm.go:73] Using DCGM to obtain processor power
I0603 10:52:04.432422       1 exporter.go:103] Kepler running on version: v0.7.12-dirty
I0603 10:52:04.432477       1 config.go:293] using gCgroup ID in the BPF program: true
I0603 10:52:04.432500       1 config.go:295] kernel version: 6.8
I0603 10:52:04.432553       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0603 10:52:04.432595       1 power.go:78] Unable to obtain power, use estimate method
I0603 10:52:04.432607       1 redfish.go:169] failed to get redfish credential file path
I0603 10:52:04.433205       1 acpi.go:71] Could not find any ACPI power meter path. Is it a VM?
I0603 10:52:04.433216       1 power.go:79] using none to obtain power
E0603 10:52:04.455263       1 dcgm.go:201] failed to get GPU Instance hierarchy: Error retrieving DCGM MIG hierarchy: Cannot perform the requested operation because NVML doesn't exist on this system.
I0603 10:52:04.457859       1 dcgm.go:451] Created device group "dev-grp-2025-06-03-10-52-04"
E0603 10:52:04.458242       1 dcgm.go:459] failed to add device 0 to group "dev-grp-2025-06-03-10-52-04": Error adding entity group type GPU, entity 0 to group: Bad parameter passed to function
E0603 10:52:04.458517       1 dcgm.go:459] failed to add device 1 to group "dev-grp-2025-06-03-10-52-04": Error adding entity group type GPU, entity 1 to group: Bad parameter passed to function
I0603 10:52:04.459207       1 dcgm.go:124] DCGM initialized successfully
I0603 10:52:04.459225       1 dcgm.go:89] Using DCGM to obtain gpu power
WARNING: failed to read int from file: open /sys/devices/system/cpu/cpu0/online: no such file or directory
I0603 10:52:04.463298       1 exporter.go:84] Number of CPUs: 24
W0603 10:52:05.022951       1 exporter.go:299] Failed to open perf event for CPU cycles: failed to open bpf perf event on cpu 0: no such file or directory
I0603 10:52:05.025469       1 watcher.go:83] Using in cluster k8s config
I0603 10:52:05.126563       1 watcher.go:229] k8s APIserver watcher was started
I0603 10:52:05.128708       1 process_energy.go:129] Using the Ratio Power Model to estimate PROCESS_COMPONENTS Power
I0603 10:52:05.128734       1 process_energy.go:130] Feature names: [bpf_cpu_time_ms bpf_cpu_time_ms bpf_cpu_time_ms   gpu_compute_util]
I0603 10:52:05.130634       1 process_energy.go:129] Using the Ratio Power Model to estimate PROCESS_TOTAL Power
I0603 10:52:05.130652       1 process_energy.go:130] Feature names: [bpf_cpu_time_ms]
I0603 10:52:05.138306       1 regressor.go:276] Created predictor linear for trainer: "SGDRegressorTrainer"
I0603 10:52:05.138326       1 model.go:125] Requesting for Machine Spec: &{genuineintel intel_xeon 24 1 167 2200 2}
I0603 10:52:05.138341       1 node_platform_energy.go:53] Using the Regressor/AbsPower Power Model to estimate Node Platform Power
I0603 10:52:05.145746       1 regressor.go:276] Created predictor linear for trainer: "SGDRegressorTrainer"
I0603 10:52:05.145785       1 regressor.go:276] Created predictor linear for trainer: "SGDRegressorTrainer"
I0603 10:52:05.145794       1 regressor.go:276] Created predictor linear for trainer: "SGDRegressorTrainer"
I0603 10:52:05.145801       1 regressor.go:276] Created predictor linear for trainer: "SGDRegressorTrainer"
I0603 10:52:05.145807       1 model.go:125] Requesting for Machine Spec: &{genuineintel intel_xeon 24 1 167 2200 2}
I0603 10:52:05.145824       1 node_component_energy.go:57] Using the Regressor/AbsPower Power Model to estimate Node Component Power
I0603 10:52:05.145898       1 prometheus_collector.go:95] Registered Container Prometheus metrics
I0603 10:52:05.145927       1 prometheus_collector.go:100] Registered VM Prometheus metrics
I0603 10:52:05.145941       1 prometheus_collector.go:104] Registered Node Prometheus metrics
I0603 10:52:05.147504       1 exporter.go:194] starting to listen on 0.0.0.0:9102
I0603 10:52:05.147620       1 exporter.go:208] Started Kepler in 715.250881ms
E0603 10:52:08.165797       1 dcgm.go:201] failed to get GPU Instance hierarchy: Error retrieving DCGM MIG hierarchy: Cannot perform the requested operation because NVML doesn't exist on this system.
E0603 10:52:08.167664       1 dcgm.go:372] failed to get running processes: Uninitialized
E0603 10:52:08.167689       1 dcgm.go:372] failed to get running processes: Uninitialized
```
It reports Metrics with DCGM source but all with zero values.
DCGM exporter is running on the same system and works fine.
```
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.183.06             Driver Version: 535.183.06   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA A100-SXM4-40GB          Off | 00000000:00:04.0 Off |                    0 |
| N/A   40C    P0              91W / 400W |  38189MiB / 40960MiB |    100%      Default |
|                                         |                      |             Enabled* |
+-----------------------------------------+----------------------+----------------------+
|   1  NVIDIA A100-SXM4-40GB          Off | 00000000:00:05.0 Off |                    0 |
| N/A   38C    P0              81W / 400W |  38189MiB / 40960MiB |    100%      Default |
|                                         |                      |             Enabled* |
+-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A    426295      C   /opt/conda/bin/python                     38180MiB |
|    1   N/A  N/A    426296      C   /opt/conda/bin/python                     38180MiB |
+---------------------------------------------------------------------------------------+
```
kepler:
  enabled: true
  daemonset:
    enabled: true
  securityContext:
    privileged: true
  image:
    repository: quay.io/sustainable_computing_io/kepler
    tag: "release-0.7.12-dcgm"
    pullPolicy: Always
  service:
    type: ClusterIP
    port: 9102
  podAnnotations:
    prometheus.io/scrape: "true"
    prometheus.io/port: "9102"
  extraEnvVars:
    KEPLER_LOG_LEVEL: "1"
    ENABLE_GPU: "true"
    ENABLE_QAT: false
    ENABLE_EBPF_CGROUPID: true
    EXPOSE_HW_COUNTER_METRICS: true
    EXPOSE_IRQ_COUNTER_METRICS: true
    EXPOSE_CGROUP_METRICS: true
    CGROUP_METRICS: '*'
    LD_LIBRARY_PATH: "/usr/local/nvidia/lib64"
    PATH: "/usr/local/nvidia/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"
  extraHostVolumes:
    - name: nvidia-install-dir-host
      hostPath: /home/kubernetes/bin/nvidia
    - name: vulkan-icd-mount
      hostPath: /home/kubernetes/bin/nvidia/vulkan/icd.d
  extraVolumeMounts:
    - name: nvidia-install-dir-host
      mountPath: /usr/local/nvidia
      readOnly: true
    - name: vulkan-icd-mount
      mountPath: /etc/vulkan/icd.d
      readOnly: true

Note when using v0.8.0-dcgm image I got log that ENABLE_GPU is false even I made it true.

conquerorAlex, 2025-07-18:
when to support GPU power?

mohamedosama113, 2025-07-29:
@conquerorAlex I am sorry I don't get you