sustainable-computing-io/kepler #2205: When to support other OS, such as Ubuntu (deb packge) in releases? Now only rpm package supported.

conquerorAlex, 2025-07-08:
Is there a plan to support other OS package for setting up, such as deb package in Ubuntu? 
At present, only rpm package in releases. 
It's an urgent need to use in other OS environment.

sthaha, 2025-07-08:
@conquerorAlex,  we currently don't have plans to support deb (or even rpm for that matter). We may remove rpms from the releases from next release onwards until we have the bandwidth to maintain these packages. 

If you have spare bandwidth and can continue to support deb packages, please feel free to raise a pull request to the reboot branch which will replace `main` in the next release.

SamYuan1990, 2025-07-08:
> [@conquerorAlex](https://github.com/conquerorAlex), we currently don't have plans to support deb (or even rpm for that matter). We may remove rpms from the releases from next release onwards until [...]

or we have just a binary file in each release, so people can download the binary and config it into service or k8s node service by their own?

sthaha, 2025-07-08:
@conquerorAlex  I am going to close this issue since the specific `deb` package need won't be addresses any time soon. But I totally agree with @SamYuan1990 , we should create binary that users can download (cc: @vprashar2929 )

closed 2025-07-08 by sthaha