sustainable-computing-io/kepler #2211: No rule to make target 'build-manifest'

conquerorAlex, 2025-07-09:
[deploying-kepler-on-a-local-kind-cluster](https://sustainable-computing.io/installation/kepler/#deploying-kepler-on-a-local-kind-cluster)

Followed the setup instuctions as above, "make build-manifest", 
but got error: No rule to make target 'build-manifest';

[image]

Strange, why cannot find build-manifest in makefile?

ExplorerRay, 2025-07-10:
I just found that the documents are marked deprecated...

https://github.com/sustainable-computing-io/kepler-doc/pull/172

sthaha, 2025-07-11:
@conquerorAlex thanks for reporting. Like @ExplorerRay pointed `build-manifests` don't exist anymore but you can find sample manifests under - https://github.com/sustainable-computing-io/kepler/tree/main/manifests/k8s 

Also please checkout - https://github.com/sustainable-computing-io/kepler/blob/main/Makefile#L149-L180

ExplorerRay, 2025-07-11:
What is the relationship between https://github.com/[sustainable-computing-io/kepler-helm-chart](https://github.com/sustainable-computing-io/kepler-helm-chart) with this stuff?

Aren't they meant for the same purpose (Use helm to install Kepler)?

sthaha, 2025-07-11:
@ExplorerRay , good question :-)

The problem this is trying to solve are 
* keep helm charts in sync with kepler 
* have an artifact you can install with every release

Keeping it a separate repo makes it hard to force contributors to keep the manifests in sync.

so if this mechanism works fine, we can archive / freeze the helm-chart repo and works until 0.9.0 version.

ExplorerRay, 2025-07-11:
I need to say that helm repo really makes the whole install process much easier. 

Why helm-chart repo needs to be archived? Could you just mark another tag or release for it?

sthaha, 2025-07-11:
> I need to say that helm repo really makes the whole install process much easier.

Because I don't use helm, could you please help me understand how it the install process easier?