sustainable-computing-io/kepler #2262: /sys/class/powercap missing

rothgar, 2025-08-05:
### Kepler Version

0.10.0 or later (Current/Supported)

### Bug Description

I just tried to deploy kepler on top of [talos linux](https://talos.dev) (I work there) and it fails with the following error.

```
failed to initialize service rapl: failed to read rapl zones: unable to read class/powercap: open /host/sys/class/powercap: no such file or directory
```
I'm curious if it's documented what creates the powercap file in the Linux kernel. We may be missing settings in our builds that I'd like to add to enable this functionality.

I'm running this on bare metal intel i7 processors (I don't remember the exact model) so it's possible my processors don't support this.

### Steps to Reproduce

1. [deploy talos](https://www.talos.dev/v1.10/introduction/getting-started/) to bare metal
2. deploy kepler with helm chart
3. 

### Expected Behavior

Daemonset pods would run

### Environment

- OS Talos 1.10
- Kubernetes 1.33.3
- containerd 2.0.5
- Intel® Tiger Lake UP3 Processor

### Logs and Error Messages

```shell
time=2025-08-05T00:09:11.341Z level=INFO source=cmd/kepler/main.go:91 msg="Loading configuration file" path=/etc/kepler/config.yaml                                                                                                                                             
time=2025-08-05T00:09:11.342Z level=INFO source=cmd/kepler/main.go:99 msg="Completed loading of configuration file" path=/etc/kepler/config.yaml                                                                                                                                
time=2025-08-05T00:09:11.342Z level=INFO source=cmd/kepler/main.go:69 msg="Kepler version information" version=v0.10.2-18-g2b565857 buildTime=2025-08-03_15:11:30 gitBranch=main gitCommit=2b565857 goVersion=go1.23.11 goOS=linux goArch=amd64                                 
                                                                                                                                                                                                                                                                                
Configuration                                                                                                                                                                                                                                                                   
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━                                                                                                                                                                                                              
log:                                                                                                                                                                                                                                                                            
    level: debug                                                                                                                                                                                                                                                                
    format: text                                                                                                                                                                                                                                                                
host:                                                                                                                                                                                                                                                                           
    sysfs: /host/sys                                                                                                                                                                                                                                                            
    procfs: /host/proc                                                                                                                                                                                                                                                          
monitor:                                                                                                                                                                                                                                                                        
    interval: 5s                                                    
    staleness: 500ms                                                
    maxTerminated: 100                                              
    minTerminatedEnergyThreshold: 10                                                                                                    
rapl:                                                               
    zones: []                                                       
exporter:                                                           
    stdout:                                                         
        enabled: false                                              
    prometheus:                                                     
        enabled: true                                               
        debugCollectors:                                            
            - go                                                    
        metricsLevel:                                               
            - node                                                  
            - process                                               
            - container                                             
            - vm                                                    
            - pod                                                   
web:                                                                
    configFile: ""                                                  
    listenAddresses:                                                
[... 7 lines cut ...]
        zones: []                                                   
kube:                                                               
    enabled: true                                                   
    config: ""                                                      
    nodeName: blah                                                  

━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
time=2025-08-05T00:09:11.342Z level=DEBUG source=cmd/kepler/main.go:125 msg="Creating all services"
time=2025-08-05T00:09:11.342Z level=DEBUG source=cmd/kepler/main.go:198 msg="Creating Prometheus exporter"
time=2025-08-05T00:09:11.342Z level=INFO source=internal/service/initializer.go:31 msg="Initializing service" service=podInformer
W0805 00:09:11.342667    6004 client_config.go:659] Neither --kubeconfig nor --master was specified.  Using the inClusterConfig.  This might not work.
time=2025-08-05T00:09:11.347Z level=DEBUG source=k8s/pod/pod.go:153 msg="setupManager with cache" service=podInformer
time=2025-08-05T00:09:11.347Z level=INFO source=k8s/pod/pod.go:131 msg="pod informer initialized" service=podInformer
time=2025-08-05T00:09:11.347Z level=INFO source=internal/service/initializer.go:31 msg="Initializing service" service=resource-informer
time=2025-08-05T00:09:11.347Z level=INFO source=internal/resource/informer.go:162 msg="Resource informer initialized successfully" service=resource-informer
time=2025-08-05T00:09:11.347Z level=INFO source=internal/service/initializer.go:31 msg="Initializing service" service=rapl
time=2025-08-05T00:09:11.347Z level=INFO source=internal/service/initializer.go:43 msg="Shutting down initialized services"
time=2025-08-05T00:09:11.347Z level=DEBUG source=internal/service/initializer.go:47 msg="skipping service shutdown" service=podInformer reason="service does not implement Shutdowner"
time=2025-08-05T00:09:11.347Z level=DEBUG source=internal/service/initializer.go:47 msg="skipping service shutdown" service=resource-informer reason="service does not implement Shutdowner"
time=2025-08-05T00:09:11.347Z level=ERROR source=cmd/kepler/main.go:54 msg="failed to initialize services" error="failed to initialize service rapl: failed to read rapl zones: unable to read class/powercap: open /host/sys/class/powercap: no such file or directory"
```

### Additional Context

I also added drivers for intel ucode and ice-firmware to see if there was something missing in the OS but I'm not sure.

rothgar, 2025-08-05:
Talos doesn't have an interactive shell environment to run commands but I can check for kernel modules and files. Let me check with the latest beta and see if it's there.

KaiyiLiu1234, 2025-08-06:
It looks to me like RAPL hardware is available but the talos os does not expose it at /host/sys. Do boot debian and see if you can see the powercap directory and see if you can deploy kepler.

rothgar, 2025-08-06:
I confirmed it's available on Debian. Now I just need to figure out what kernel settings create those files.

[image]

sthaha, 2025-08-07:
@rothgar @KaiyiLiu1234 , so the way this is supposed to work is that the hosts' /sys and /proc should be mounted readonly to kepler pod as `/host/sys` and `/host/proc` instead of mounting it as kepler containers `/sys` and `/proc` 

Your host (Baremetal) should have powercap available .. 
You can verify this by `sudo cat /sys/class/powercap/intel-rapl:0/energy_uj` . E.g. 

```
❯  sudo cat /sys/class/powercap/intel-rapl:*/{name,energy_uj}
core
package-0
34622206960
35530206863
```

if the host has powercap available, mounting the host's `/sys` and `/proc` to containers `/host/sys` and `/host/proc` as **readonly** is the recommended way to deploy kepler on k8s.

When deploying Kepler on K8s, follow these guidelines for accessing host resources:

### Recommended Approach
1. Mount the host's `/sys` and `/proc` filesystems as **readonly** volumes under:
   - `/host/sys` (for sys)
   - `/host/proc` (for proc)
2. This approach provides secure access to host system information without compromising container isolation

### Verifying Powercap Availability
If your bare metal hosts have powercap functionality enabled, you can verify with:

```bash
❯  sudo cat /sys/class/powercap/intel-rapl:*/{name,energy_uj}
core
package-0
34622206960
35530206863
```

### Why This Configuration?
Mounting `/sys` and `/proc` as read-only under a dedicated path like `/host/` ensures that Kepler containers have the necessary system information while maintaining security best practices for Kubernetes deployments.

rothgar, 2025-08-08:
I don't have a problem with the kepler deployment. My host doesn't have the `/sys/class/powercap/*` files and I'm trying to figure out what kernel settings are needed to enable that. We try to keep Talos Linux stripped down to only what is needed and I'm looking for information on what kernel settings or drivers enable the powercap directory to be populated.

I have a test kernel that enables powercap and this created the powercap directory but no other files under it.

We enabled these on the Linux kernel kernel. FWIW we're using the upstream LTS 6.12.40 kernel.
```
+CONFIG_INTEL_TPMI_POWER_DOMAINS=m
+CONFIG_INTEL_TPMI=m
+CONFIG_INTEL_PLR_TPMI=m
+CONFIG_POWERCAP=y
+CONFIG_INTEL_RAPL_CORE=m
+CONFIG_INTEL_RAPL=m
+CONFIG_INTEL_RAPL_TPMI=m
```

sthaha, 2025-08-12:
@rothgar please note for MSR support, I am adding a proposal - https://github.com/sustainable-computing-io/kepler/pull/2271

rothgar, 2025-08-12:
Sure, you can let me know when something is ready to be tested.

closed 2026-07-08 by rothgar