sustainable-computing-io/kepler #2282: API endpoints for health(z)

sthaha, 2025-08-17:
### Target Version

Current version (0.10.0+) - New Architecture

### Feature Description

Applications running in Kubernetes often rely on external probes (liveness/readiness) to determine pod health. Currently, the kepler does not expose any HTTP endpoints that can be used by these probes or by monitoring systems. This makes it difficult to:

1. Verify that the service is fully initialized before accepting traffic.
2. Detect and recover from runtime failures automatically via Kubernetes’ self‑healing mechanisms.

### Desired Feature

### Problem Statement

Currently, the kepler does not expose any HTTP endpoints that can be used by these probes or by monitoring systems. This makes it difficult to:

1. Verify that the service is fully initialized before accepting traffic.
2. Detect and recover from runtime failures automatically via Kubernetes’ self‑healing mechanisms.

### Proposed Solution

Introduce two lightweight RESTful endpoints: (see: https://kubernetes.io/docs/reference/using-api/health-checks/ )

| Endpoint | Purpose | Example Response |
|----------|---------|------------------|
| `GET /healthz` | **Readiness** – confirms the application is ready to serve traffic. | `{ "status": "ok" }` |
| `GET /livez` | **Liveness** – checks that the process is running and not dead‑locked. | `{ "status": "alive" }` |

Both endpoints should:

- Return HTTP 200 when healthy.
- Return a non‑2xx status (e.g., 503) when unhealthy.

sthaha, 2025-08-20:
@ExplorerRay thats only an example .. I think a better example would have been https://kubernetes.io/docs/concepts/configuration/liveness-readiness-startup-probes/

There are readiness, and liveness probes and the actual endpoint really doesn't matter but it is better to follow the convention. so /healthz and /livez are just fine. Perhaps the path could be 
* /probe/readyz
* /probe/livez

ExplorerRay, 2025-08-21:
Sorry for mistaking that. I originally thought it is something like a standard, but it is just an implementation of Kubernetes API server.

robinlem, 2025-09-09:
Hi All! I’ve implemented the endpoints requested in this issue:

This is my first PR to Kepler—feedback is very welcome, and I’m happy to adjust anything I missed

**Endpoints created**

/probe/readyz: Readiness probe (200 after the first sample has been collected)

/probe/livez: Liveness probe (200 if sampling has been active within the last 30s)

**Performance**

< 1 ms response time (vs 50–200 ms for /metrics)

Minimal CPU/memory usage

JSON responses as specified

**Files changed**

internal/server/probe.go — endpoint implementation

internal/server/probe_test.go — comprehensive tests

internal/monitor/monitor.go — added LastCollectionTime()

manifests/helm/kepler/ — Helm configuration with legacy fallback

HEALTH_PROBES.md — full documentation

**Helm configuration**

```
daemonset:
  useOptimizedProbes: true  # default
  livenessProbe:
    httpGet:
      path: /probe/livez
  readinessProbe:
    httpGet:
      path: /probe/readyz
```

Backward compatibility: set `useOptimizedProbes: false `to fall back to /metrics.

Ready for review!