sustainable-computing-io/kepler #2302: Publish helm repo to github oci section

jcpunk, 2025-08-27:
### Target Version

Current version (0.10.0+) - New Architecture

### Feature Description

Modern versions of helm are able to pull charts from an OCI repo.  The github container registry is already used by a number of projects for this purpose.

### Problem Statement

Currently the helm install instructions point to an assembled tarball that doesn't work with all helm abstraction layers.

### Proposed Solution

https://github.com/marketplace/actions/helm-publish

### Alternatives Considered

Setting up my own hacky repo mirror

### Additional Context

https://helm.sh/docs/topics/registries/
https://github.com/prometheus-community/helm-charts/tree/main/charts/kube-prometheus-stack#install-helm-chart

sunya-ch, 2025-09-17:
@ExplorerRay shall this issue closed by https://github.com/sustainable-computing-io/kepler/pull/2324?

ExplorerRay, 2025-09-17:
The current state of `oci://quay.io/sustainable_computing_io/charts/kepler` is still unavailable because it requires a release to trigger that CI.

I think whether this issue can be closed depends on @jcpunk. Maybe we need another CI for pushing helm chart to registry when merging to main in addition to only release. Or some lines of docs to tell users that we can use this way to install kepler helm chart.